arduino/Arduino #6615: I2C Slave Transmit ISR is incorrectly setting ACK

petertorelli, 2017-08-15:
According to the TI description of I2C protocol (http://www.ti.com/lit/an/slva704/slva704.pdf page 5, sect 2.3, bullet 4) the receiver controls ACK/NACK, not that transmitter.

The fall-through case `TW_ST_DATA_ACK` in `twi.c` in the ISR is setting ACK/NACK based on the state of the 32-byte buffer (have all bytes been written). I believe this is backwards based on the spec, the ST code should continue sending bytes from the buffer until the master-receiver asserts NACK followed by STOP (page 6, figure 7 of the same doc).

I encountered this issue trying to interface with a master platform that is using a DMA engine to read a variable number of bytes from the I2C slave (16 to 1024). Unless the slave knows exactly how many bytes to send, the master has to wake up and generate another interrupt.

matthijskooijman, 2017-08-18:
I think you're talking about this code: https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/libraries/Wire/src/utility/twi.c#L534-L550 is that correct? I do not see any fallthrough in the `TW_ST_DATA_ACK` case there, only for the `TW_ST_DATA_NACK` case?

If I read the datasheet correctly, in ST mode after transmitting a data byte, te `TWEA` bit (set by `twi_reply()` does not indicate whether the slave should *transmit* an ACK, but whether to *expect* an ACK. After sending the last byte from the buffer, it signals to expect a NACK. If the master sends an ACK regardless, the TW_ST_LAST_DATA event is triggered, making the slave fall back to the idle state. You say "the ST code should continue sending bytes from the buffer", but there are no bytes to send when the end of the buffer is reached. The slave could send arbitrary bytes, but by dropping off the bus, the master will just read 0xff bytes (from the pullup), which is just as correct.

I'm not sure about your problem with your DMA engine, but I can imagine that problem has some other cause? Or perhaps I completely misunderstood your point?

petertorelli, 2017-08-18:
@matthijskooijman 

Thanks for clarifying `twi_reply()`, that was part of my confusion.

The slave falling back to idle state after receiving ACK when the buffer is empty is my issue. The DMA device is readying 0xFF because the Slave went into idle mode, because the Arduino implementation is limited to 32B per start transaction, period.

But wouldn't it make more sense to call the user-space `onRequest` (via `twi_onSlaveTransmit`) if the master is still ACK'ing and let the user decide whether to refill the buffer, rather than put the device in idle? There's no limit as to how many bytes I2C ST can send, but the code is forcing it to 32, which is why the author of the master's code is complaining: having to wakeup every 32B is hurting their power.

How about this: in `TW_ST_DATA_ACK` case, if the buffer drains (`twi_txBufferIndex >= twi_txBufferLength`), what if the those counters are reset and `twi_onSlaveTransmit` is called again, and remove the `twi_reply()`? Problem is I'm not confident this won't break other usage models.

petertorelli, 2017-08-19:
Here's what I changed it to, and it works for my case, but I don't know what you use for general unit testing and regression, so I can't quite make a pull request.

~~~
    case TW_ST_DATA_ACK: // byte sent, ack returned
      // copy data to output register
      TWDR = twi_txBuffer[twi_txBufferIndex++];
      twi_reply(1); //always ACK

      // If the buffer drained, call the user callback to refill it
      if (twi_txBufferIndex >= twi_txBufferLength){
        twi_txBufferIndex = 0;
        twi_txBufferLength = 0;
        twi_onSlaveTransmit();
      }
/*      
      // if there is more to send, ack, otherwise nack
      if (twi_txBufferIndex < twi_txBufferLength){
        twi_reply(1);
      }else{
        twi_reply(0);
      }
*/
      break;
~~~

matthijskooijman, 2017-08-21:
Ah, then this is more of a feature request than a bug report :-)

The solution you propose would probably work, though it requires modifying the `onRequest` API to indicate whether this is the first request or not (or, more generically, perhaps how many bytes were previously sent as part of this read transaction). I'm not sure how to approach this exactly, though.

petertorelli, 2017-08-22:
@matthijskooijman ...

Yes, I guess it is a feature request. :) 

I've solved my problem and a Master Reader can now read an arbitrary number of bytes from a Slave Transmitter, not just 32, per a single I2C request.

Lacking any regression testing I won't generate a pull request, but I'll close this and perhaps future generations may unearth it.

Thanks for the help!

closed 2017-08-22 by petertorelli