arduino/Arduino #6616: [Library Manager] arduino library for TS4231 Light to Digital Converter

jseibel6900, 2017-08-15:
Please add the TS4231 library to the library manager list.  I think I have met all of the 1.5 format requirements, but this is my first library so please let me know if I missed something.  Thanks.

[https://github.com/TriadSemi/TS4231]

jseibel6900, 2017-08-16:
Hi per1234 - 
Do I need to do anything else to have the library added to the library manager list?  The library manager FAQ says the list is updated about every hour, but I don't know if the library has officially been accepted so that it can be added to the list.  I'm sure I'm missing something.  Please let me know how to proceed.
Thanks!

per1234, 2017-08-16:
It looks like your library meets all the requirements for addition to Library Manager. The way it works is that cmaglie comes through every few days and adds all the libraries that have been submitted as long as they don't have any issues that prevent this. Once that is done, the libraries should be added to Library Manager via the hourly job. There [may be a longer delay](https://github.com/arduino/Arduino/issues/6593#issuecomment-321559057) before they are added to the Arduino Web Editor's Library Manager, I'm not sure whether that is still the case.

So there's nothing more you have to do, just wait for the notification that cmaglie has closed this issue and then check Library Manager in the standard IDE a couple hours later to be sure that your library was successfully added.

jseibel6900, 2017-08-17:
Got it.  Thanks for the quick response!

cmaglie, 2017-08-17:
Done!

closed 2017-08-17 by cmaglie