arduino/Arduino #6621: [Library Manager] Please add cQueue library

SMFSW, 2017-08-16:
Please add cQueue library to the library manager:
c port of Queue library with needed changes/refactoring

https://github.com/SMFSW/cQueue

thanks

cmaglie, 2017-08-17:
Done!

closed 2017-08-17 by cmaglie

SMFSW, 2019-12-01:
I was reported an issue with the latest update of cQueue library by the Arduino library manager.

The latest available version to download is denoted v1.6, but it's content is v1.5.
The library properties file is mentioning v1.6, but it seems to be the only updated file.
Also, V1.5 is not mentioned for download in the dropdown list.

I checked my library tag in github and appears to be correct.
Can you have a check? Thanks.

[https://github.com/SMFSW/cQueue/issues/1](https://github.com/SMFSW/cQueue/issues/1)

per1234, 2019-12-02:
@SMFSW the problem is that the `version` value in library.properties was set to `1.6` before you created your `v1.5` tag:
https://github.com/SMFSW/cQueue/blob/v1.5/library.properties#L2
```
version=1.6
```
Then the `version` value was left unchanged for the `v1.5` tag:
https://github.com/SMFSW/cQueue/blob/v1.6/library.properties#L2

The Library Manager indexer job detects new releases using Git tags, but it does its versioning using the `version` field of library.properties. So since it already had a `1.6` version in the index (added by the `v1.5` tag), the `v1.6` tag was ignored.

To fix this, you'll either need to create a new tag with an updated `version` value, or else redo the existing tag(s).

SMFSW, 2019-12-02:
oops, my mistake.
Sorry, I did just check about v1.6 tag, I didn't thought I made a mistake in the previous tag.

Thanks!