arduino/Arduino #6631: MKR1000 port not showing on IDE, MacOS X

dackdel, 2017-08-17:
I did the double reset that I read from the previous thread. But still have the same issue. The double reset made it work once but thats it.

facchinm, 2017-08-17:
Hi @dackdel ,
where did you find the "double reset" info? just the link the important parts of the conversation.
In fact, the double click allows the board to be in bootloader mode forever, with no timeout. You should then upload a sketch (a blink, for example) to reacquire the complete functionality. 
Let me know if it work for you.

dackdel, 2017-08-17:
https://github.com/arduino/Arduino/issues/5223 i found it here

facchinm, 2017-08-17:
Thanks for the link! Are you able to upload a sketch after the double tap?

dackdel, 2017-08-17:
no

facchinm, 2017-08-17:
Can you paste here the "verbose" output of the upload? you can enable it under File/Preferences -> verbose output for compilation and upload

dackdel, 2017-08-17:
ok it started working again i aggressively kept pressing reset. left the lab now. will post the verbose tomo?

cmaglie, 2017-08-29:
If the board is working there is no more reason to post the verbose output.
I'm closing this as solved, please reopen if you have other questions.

closed 2017-08-29 by cmaglie