arduino/Arduino #6634: New OPC Library v2.0 for Arduino/Genuino released

ildemartinez, 2017-08-18:
Hello

I am proud to release the OPC library for Arduino. It is widly used but I would like to be available using the Arduino IDE library manager. I made a releas a few hours ago but the library manager does not detect the update. I think the library.properties file is ok but not sure.

Here is the url of the release 2.0:

https://github.com/SoftwareTools4Makers/OPC/releases/tag/2.0

The main project page is at: http://www.st4makers.com/

All the best
ilde

PaulStoffregen, 2017-08-18:
Just an related comment... you might consider the paragraph line from library.properties (and maybe the readme) is likely to be seen by Arduino users who have no idea what OPC is about.

per1234, 2017-08-18:
@ildemartinez the `version` value in your library.properties must match the release. Since the version value was changed after the release (https://github.com/SoftwareTools4Makers/OPC/commit/3a04b56a00e7825539a1d5f444566662364970aa) this is currently not the case with your library and also is not the case with any of the previous releases.

Please either create a new release or tag with this problem resolved or redo the current release and then comment here to let us know that we can proceed with adding this library to the Library Manager index.

ildemartinez, 2017-08-19:
Oooops!!!

sorry...tagged before commit and release!!

I think is solved!!

https://github.com/SoftwareTools4Makers/OPC/releases/tag/2.0

May you test it? Thank you a lot!! 

all the best
ilde

ildemartinez, 2017-08-19:
I see the ver 2.0 from the Arduino IDE now!! seems to be ok!

thank you for the great support...next time I will do it better

all the best
ilde

per1234, 2017-08-19:
I see now. I hadn't realized the library had been added to the Library Manager index previously (https://github.com/arduino/Arduino/issues/3843) and this issue was only about the new release not being picked up by the Library Manager job.

I originally thought this was a request for the library to be added to the index, which was why I reopened it.

Closing as solved.

closed 2017-08-19 by per1234