arduino/Arduino #6635: TridentTD Esp8266 Simple-Pair library release V1.0

TridentTD, 2017-08-18:
Hello

I have just released the ESP8266 Simple-Pair library for Arduino. 
Used for checking hand between ESP8266s to exchange some data 
before WiFi connection, such as  SSID, PASSWORD or ESP-NOW KEY ..etc.
And can use these data to connect WiFi after. 

https://github.com/TridentTD/TridentTD_SimplePair/releases/tag/1.0

Thank you.

cmaglie, 2017-08-21:
done!

closed 2017-08-21 by cmaglie

HobbytronicsPK, 2017-10-08:
Can I use your library with some android app (I'm thinking to develop one) to connect ESP8266 to existing WiFi connection with the help of smart phone? Will it work the same way, as Itead Sonoff device do for pairing to existing WiFi network?

I use my [ESPMetRED ](https://github.com/HobbytronicsPK/ESPMetRED)Library to communicate ESP8266 with Nodered, and want to incorporate WiFi SSID, Password, MQTT Credentials via smart phone. 

If it is possible, may you help in to write an example sketch?

per1234, 2017-10-08:
@HobbytronicsPK this is not an appropriate place to request support for this library.