arduino/Arduino #6636: IDE unresponsive when moved to secondary display rotated 90deg.

alangbeen, 2017-08-18:
Running IDE version 1.8.3 on Windows7 Pro.  Running 3 monitors: the primary (laptop) and one secondary in normal orientation, third is rotated 90 deg.  The IDE becomes unresponsive if moved to the rotated display.  When moved back to a horizontal display, it responds normally again.  
The IDE will only work on the rotated display when it is configured as the 'primary' display in Windows.  This is not a desirable configuration for a laptop that is regularly disconnected and used elsewhere.

feikname, 2017-08-21:
@alangbeen Is the Arduino IDE the only software that has performance problems on this rotated display? Could you please try opening [Processing](https://processing.org/) on it too?

facchinm, 2017-09-04:
As @feikname correctly states, this is probably related with and incompatibility between Java and Windows, so Processing (and Eclipse) would behave in the same way. Could you test with one of these software and report back? Thanks!

facchinm, 2017-09-12:
Closing as duplicate of #6600, keeping track there

closed 2017-09-12 by facchinm