arduino/Arduino #6643: Adjust wiring_analog.c for Attiny841/441

mschlenker, 2017-08-21:
Attiny841/44 with 12+5 ADCs use ADMUXA and ADMUXB instead of ADMUX to select analog reference and channel. Just look at hardware/tools/avr/avr/include/avr/iotn841.h and hardware/tools/avr/avr/include/avr/iotn441.h.

The patch attached properly initializes ADMUXA and ADMUXB to make ADC usable, tested with Attiny841, thanks @awatterott for the hint. 

[wiring_analog_attiny841.txt](https://github.com/arduino/Arduino/files/1238731/wiring_analog_attiny841.txt)

per1234, 2017-08-21:
If you're proposing a code change it would be better to submit a pull request.

mschlenker, 2017-08-21:
+per1234, thank you for the hint:
https://github.com/arduino/Arduino/pull/6644