arduino/Arduino #6646: Library manager should display license for each library

scls19fr, 2017-08-22:
Hello,

Library manager should display license for each library.
Ideally we should find SPDX identifier https://spdx.org/licenses/ https://en.wikipedia.org/wiki/Software_Package_Data_Exchange for each library that can be installed using Arduino IDE.

I didn't find in
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification
anything related to license

Pinging @KuangLei @cmaglie @per1234

Kind regards

scls19fr, 2017-08-22:
This is probably out of the scope of this issue but I still wonder why 2 configuration files: `library.json` and `library.properties`
Why not just one (ideally [Yaml](https://en.wikipedia.org/wiki/YAML))?

A GitHub bot sending Pull Requests may be considered for libraries (if many Arduino libraries are on GitHub) - An exemple (in a different context) can be found at https://juliacomputing.com/blog/2017/08/17/femtocleaner.html https://github.com/JuliaComputing/FemtoCleaner.jl

KuangLei, 2017-08-23:
@scls19fr Only one configuration file, `library.properties`, is required for making a library. See the [Library Manager FAQ](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ).  

For all the existing libraries, I suggest to collect the license through:  

* checking the `LICENSE` or `LICENSE.md` file with the [Ruby Gem](https://github.com/benbalter/licensee) mentioned above or any other tools,
* or emailing the maintainer (or the author in `library.properties`) to add a license.

For the existing library that can't be found or added a license, I suggest to remove it.  

Thanks.

per1234, 2017-08-23:
library.json is used by platformio. It's not used at all by the Arduino IDE/Library Manager.

scls19fr, 2017-08-23:
Thanks @per1234 about `library.json` usage. I thought it was related to Arduino IDE.
@KuangLei I really think that adding a field for license to `library.properties` could be a good idea.
If this field is not present than user will have to dive into library repository. Library manager could, in such a case, simply indicate:

- `library.properties` doesn't provide a license field. Check the `LICENSE` or `LICENSE.md` inside library repository 

(or something similar)

Respecting license (open source or not) is crucial and when choosing a library, license is an important criterion to consider.

It will make Arduino users (especially new users) aware of this.

KuangLei, 2017-08-23:
@scls19fr I agree with you of adding a field for license in the `library.properties`.  
Sorry for my bad English, and what I meant before was:
> For all the **existing** libraries, I suggest to **collect** the license (and then storage the license in the library server) through:...

scls19fr, 2017-08-23:
My english is not perfect also ;-) so I may have misunderstood.
For existing libraries, if license can't be find easily I think Arduino library manager should simply warm user about the current legal situation. (Not to remove the library immediately).

A first step could be to edit library specification
and quickly contact library developers (email, blog post, ...) about this new field.
Measure how many libraries have this field set (after some weeks / months)
When this field reach a quite high usage display it into library manager (or a warning when it's not set).
Contact again developers that didn't update (some months / year later for example) to update their `library.properties` and publish a new tag.

KuangLei, 2017-08-23:
What I meant above is not to remove the library immediately, sorry.  
But if months later for example, the maintainer still doesn't provide a license, I think the library should be removed.

per1234, 2017-08-23:
It's not clear to me what the intent is behind this suggestion. Here are some comments I have on each component:

### Add a license field to library.properties
I'm not necessarily against this but it just seems like the sort of user who is concerned with the license type will have no trouble looking for it in the library folder/source code.

### Display the license value in Library Manager
The average Arduino user doesn't care about licenses and probably doesn't even understand why they should. The Arduino IDE attempts to offer a very simple UI to be beginner friendly so we need to carefully evaluate whether adding additional complexity is justified.

### Add a warning if the license field of library.properties is not supplied
As someone who has been very active in trying to get library authors to fix this sort of warning I can tell you they have the side effect of causing confusion and unnecessary concern for users and are not always effective at getting library authors to add/fix their metadata.

### Eventually add licensing requirements for inclusion in the Library Manager index
Is this coming from a legal perspective (i.e. it's not legal for Arduino to distribute all rights reserved libraries via Library Manager) or solely to advocate for open source licensing of these libraries?

As to the former, of course that would be a concern but I don't have the knowledge to comment on this. 

As to the latter, I certainly do understand the importance of licensing. I think a lot of Arduino users just assume that any code published online can be used freely but of course this is not the case and I have seen that assumption cause problems (e.g. https://github.com/arduino/Arduino/issues/4604). A more light handed approach would be to simply add a note to the [Library Manager FAQ](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ) that recommends adding a license with a link to more information.

bperrybap, 2018-01-10:
A light handed approach might be to try to auto detect the license for a library based on its contents and display it in the library manager if it could be detected. This would be similar to what github currently does.

njh, 2018-05-25:
Looks like a `license` field has been added to [library_index.json](https://downloads.arduino.cc/libraries/library_index.json).

Is this based on the GitHub license detection?

scls19fr, 2018-05-25:
If `license` field is coming from `library.properties` and given that we have now a repository URL for many libraries thanks to https://github.com/arduino/Arduino/issues/7591, we can probably clone repositories, detect which libraries don't have a license field in `library.properties`, detect (thanks to [licensee](https://github.com/benbalter/licensee)) what license is use (when no license field in `library.properties`) and submit automatically a pull request to fix `library.properties` accordingly .

We can also detect inconsistency in license (ie a different license in `library.properties` and LICENSE file) and submit PR to inform library developer.

scls19fr, 2018-05-31:
From https://github.com/olikraus/U8g2_Arduino/pull/4#issuecomment-393621789

[Wiki](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#library-metadata) should be updated to provide what license field should be.

olikraus, 2018-05-31:
Well, fine, but is the "license" keyword already detected as per the Arduino IDE?
Or will unknown fields simply ignored?