arduino/Arduino #6649: MacOS X High Sierra - Uno Wifi Network Upload Fails

neilmw1, 2017-08-23:
Whenever I try to do a network upload to one of my Arduino Uno Wifi boards using 1.8.3 or 1.8.4 in MacOS X High Sierra, it fails with an error message.  I'm not sure if this message varies but its consistent every time - I can upload to the boards perfectly using the Windows version of the software so I have a workaround but it would be good if you could take a look at this?  

The error message after compiling and preparing to upload is as follows (it doesn't matter what I try to upload to any of my 3 Uno WiFi boards - this is the blink example) :-

the selected serial port runtime.mpreinit(0x59cf40) does not exist or your board is not connected.

`Arduino: 1.8.4 (Mac OS X), Board: "Arduino Uno WiFi"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/neil/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/neil/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/neil/OneDrive/Arduino/libraries -fqbn=arduino:avr:unowifi -ide-version=10804 -build-path /var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_150171 -warnings=none -build-cache /var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_cache_515704 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/neil/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/neil/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/neil/OneDrive/Arduino/libraries -fqbn=arduino:avr:unowifi -ide-version=10804 -build-path /var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_150171 -warnings=none -build-cache /var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_cache_515704 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
Using board 'unowifi' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Using core 'arduino' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Detecting libraries used...
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_UNO_WIFI_DEV_ED -DARDUINO_ARCH_AVR  -DESP_CH_UART -DESP_CH_UART_BR=19200 "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_150171/sketch/Blink.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_UNO_WIFI_DEV_ED -DARDUINO_ARCH_AVR  -DESP_CH_UART -DESP_CH_UART_BR=19200 "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_150171/sketch/Blink.ino.cpp" -o "/var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_150171/preproc/ctags_target_for_gcc_minus_e.cpp"
"/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_150171/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_UNO_WIFI_DEV_ED -DARDUINO_ARCH_AVR  -DESP_CH_UART -DESP_CH_UART_BR=19200 "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_150171/sketch/Blink.ino.cpp" -o "/var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_150171/sketch/Blink.ino.cpp.o"
Compiling libraries...
Compiling core...
Using precompiled core
Linking everything together...
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p  -o "/var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_150171/Blink.ino.elf" "/var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_150171/sketch/Blink.ino.cpp.o" "/var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_150171/../arduino_cache_515704/core/core_arduino_avr_unowifi_51f02b7210b938436b779d1c032618e1.a" "-L/var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_150171" -lm
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-objcopy" -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0  "/var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_150171/Blink.ino.elf" "/var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_150171/Blink.ino.eep"
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-objcopy" -O ihex -R .eeprom  "/var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_150171/Blink.ino.elf" "/var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_150171/Blink.ino.hex"
Sketch uses 928 bytes (2%) of program storage space. Maximum is 32256 bytes.
Global variables use 9 bytes (0%) of dynamic memory, leaving 2039 bytes for local variables. Maximum is 2048 bytes.
An error occurred while uploading the sketch
failed MSpanList_Insert 0x624000 0x31294daea3cb 0x0 0x0
fatal error: MSpanList_Insert

runtime stack:
runtime.throw(0x40be70, 0x10)
	/home/martino/go_src/src/runtime/panic.go:530 +0x90 fp=0x7ffeefbff5a0 sp=0x7ffeefbff588
runtime.(*mSpanList).insert(0x5a1ec8, 0x624000)
	/home/martino/go_src/src/runtime/mheap.go:933 +0x293 fp=0x7ffeefbff5d0 sp=0x7ffeefbff5a0
runtime.(*mheap).freeSpanLocked(0x5a16c0, 0x624000, 0x100, 0x0)
	/home/martino/go_src/src/runtime/mheap.go:809 +0x4be fp=0x7ffeefbff638 sp=0x7ffeefbff5d0
runtime.(*mheap).grow(0x5a16c0, 0x8, 0x0)
	/home/martino/go_src/src/runtime/mheap.go:675 +0x2a0 fp=0x7ffeefbff690 sp=0x7ffeefbff638
runtime.(*mheap).allocSpanLocked(0x5a16c0, 0x1, 0x0)
	/home/martino/go_src/src/runtime/mheap.go:553 +0x4e3 fp=0x7ffeefbff6e8 sp=0x7ffeefbff690
runtime.(*mheap).alloc_m(0x5a16c0, 0x1, 0x15, 0x0)
	/home/martino/go_src/src/runtime/mheap.go:437 +0x119 fp=0x7ffeefbff718 sp=0x7ffeefbff6e8
runtime.(*mheap).alloc.func1()
	/home/martino/go_src/src/runtime/mheap.go:502 +0x41 fp=0x7ffeefbff748 sp=0x7ffeefbff718
runtime.systemstack(0x7ffeefbff768)
	/home/martino/go_src/src/runtime/asm_amd64.s:307 +0xab fp=0x7ffeefbff750 sp=0x7ffeefbff748
runtime.(*mheap).alloc(0x5a16c0, 0x1, 0x10000000015, 0x10c5f)
	/home/martino/go_src/src/runtime/mheap.go:503 +0x63 fp=0x7ffeefbff798 sp=0x7ffeefbff750
runtime.(*mcentral).grow(0x5a32c0, 0x0)
[... cut ...]
	/home/martino/go_src/src/runtime/os1_darwin.go:140 +0x1f fp=0x7ffeefbff9f0 sp=0x7ffeefbff9d8
runtime.mcommoninit(0x59cf40)
	/home/martino/go_src/src/runtime/proc.go:494 +0x105 fp=0x7ffeefbffa38 sp=0x7ffeefbff9f0
runtime.schedinit()
	/home/martino/go_src/src/runtime/proc.go:434 +0x79 fp=0x7ffeefbffa80 sp=0x7ffeefbffa38
runtime.rt0_go(0x7ffeefbffab8, 0xf, 0x7ffeefbffab8, 0x0, 0x0, 0xf, 0x7ffeefbffc48, 0x7ffeefbffc92, 0x7ffeefbffc9b, 0x7ffeefbffcaa, ...)
	/home/martino/go_src/src/runtime/asm_amd64.s:138 +0x132 fp=0x7ffeefbffa88 sp=0x7ffeefbffa80`

neilmw1, 2017-08-23:
Also get the following 

`	/home/martino/go_src/src/runtime/proc.go:2634 +0x27 fp=0x7ffeefbff9d8 sp=0x7ffeefbff9a0
runtime.mpreinit(0x59cf40)
the selected serial port runtime.mpreinit(0x59cf40)
 does not exist or your board is not connected
	/home/martino/go_src/src/runtime/os1_darwin.go:140 +0x1f fp=0x7ffeefbff9f0 sp=0x7ffeefbff9d8
runtime.mcommoninit(0x59cf40)
	/home/martino/go_src/src/runtime/proc.go:494 +0x105 fp=0x7ffeefbffa38 sp=0x7ffeefbff9f0
runtime.schedinit()
	/home/martino/go_src/src/runtime/proc.go:434 +0x79 fp=0x7ffeefbffa80 sp=0x7ffeefbffa38
runtime.rt0_go(0x7ffeefbffab8, 0xf, 0x7ffeefbffab8, 0x0, 0x0, 0xf, 0x7ffeefbffc48, 0x7ffeefbffc92, 0x7ffeefbffc9b, 0x7ffeefbffcaa, ...)
	/home/martino/go_src/src/runtime/asm_amd64.s:138 +0x132 fp=0x7ffeefbffa88 sp=0x7ffeefbffa80`

cmaglie, 2017-08-28:
could you enable verbose upload?

It seems related to https://github.com/golang/go/issues/17182, maybe a rebuild of the tool with a recent version of golang may fix it.

neilmw1, 2017-08-28:
Hi, 

As requested

Compilation:
```
/Applications/Arduino.app/Contents/Java/arduino-builder` -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/neil/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/neil/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/neil/OneDrive/Arduino/libraries -fqbn=arduino:avr:unowifi -ide-version=10804 -build-path /var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_914572 -warnings=none -build-cache /var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_cache_583261 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/neil/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/neil/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/neil/OneDrive/Arduino/libraries -fqbn=arduino:avr:unowifi -ide-version=10804 -build-path /var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_914572 -warnings=none -build-cache /var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_cache_583261 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /Applications/Arduino.app/Contents/Java/examples/01.Basics/Blink/Blink.ino
Using board 'unowifi' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Using core 'arduino' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Detecting libraries used...
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_UNO_WIFI_DEV_ED -DARDUINO_ARCH_AVR  -DESP_CH_UART -DESP_CH_UART_BR=19200 "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_914572/sketch/Blink.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_UNO_WIFI_DEV_ED -DARDUINO_ARCH_AVR  -DESP_CH_UART -DESP_CH_UART_BR=19200 "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_914572/sketch/Blink.ino.cpp" -o "/var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_914572/preproc/ctags_target_for_gcc_minus_e.cpp"
"/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "/var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_914572/preproc/ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_UNO_WIFI_DEV_ED -DARDUINO_ARCH_AVR  -DESP_CH_UART -DESP_CH_UART_BR=19200 "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_914572/sketch/Blink.ino.cpp" -o "/var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_914572/sketch/Blink.ino.cpp.o"
Compiling libraries...
Compiling core...
"/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc" -c -g -x assembler-with-cpp -flto -MMD -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10804 -DARDUINO_AVR_UNO_WIFI_DEV_ED -DARDUINO_ARCH_AVR  -DESP_CH_UART -DESP_CH_UART_BR=19200 "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard" "/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino/wiring_pulse.S" -o "/var/folders/l1/t0yygvzd0_75n4cm4lrgk9480000gn/T/arduino_build_914572/core/wiring_puls [... cut ...]
runtime.schedinit()
	/home/martino/go_src/src/runtime/proc.go:434 +0x79 fp=0x7ffeefbffa80 sp=0x7ffeefbffa38
runtime.rt0_go(0x7ffeefbffab8, 0xf, 0x7ffeefbffab8, 0x0, 0x0, 0xf, 0x7ffeefbffc48, 0x7ffeefbffc92, 0x7ffeefbffc9b, 0x7ffeefbffcaa, ...)
	/home/martino/go_src/src/runtime/asm_amd64.s:138 +0x132 fp=0x7ffeefbffa88 sp=0x7ffeefbffa80
the selected serial port 	/home/martino/go_src/src/runtime/asm_amd64.s:138 +0x132 fp=0x7ffeefbffa88 sp=0x7ffeefbffa80
 does not exist or your board is not connected
```

facchinm, 2017-09-04:
Recompiling https://github.com/arduino/arduinoOTA with go 1.9 should be enough, assigning to NEXT milestone.

neilmw1, 2017-09-15:
I can confirm recompiling against go 1.9 worked, thanks for your help.

per1234, 2020-10-02:
It appears the updated version of ArduinoOTA has been bundled with the IDE, so I'll close this as fixed.

closed 2020-10-02 by per1234