arduino/Arduino #6652: The Wire library should get its own .readBytes() without timeout.

Koepel, 2017-08-25:
The Stream.readBytes() uses a timeout. In my opinion that does not go well with the Wire library.

The Wire library is the only library in the Stream family that does not use data as a stream. The buffer(s) are cleared and the data is used as packets of data. There is no situation that requires to wait or a timeout.

Therefore I suggest to give the Wire library its own Wire.readBytes() without the timeout.
Almost everyone uses a loop to read the data. I prefer using Wire.readBytes(), but only without the disadvantages of the timeout.

For the Wire library any timeout by accidently reading too much data will only cause a delay, it does not have a purpose. The onReceive() and onRequest() functions are run in a interrupt handler, in that case a useless delay might even influence the whole sketch.

The improved Teensy Wire library by @nox771 should be checked that it will not cause a conflict.   [github.com/nox771/i2c_t3](https://github.com/nox771/i2c_t3)

Koepel, 2017-09-07:
One solution could be that the timeout is set to zero to disable it with `Wire.setTimeout(0);` in `Wire.begin()`. As far as I know, that does not have any disadvantages.  
Thanks to: https://github.com/Lauszus/BalancingRobotFullSize/issues/2#issuecomment-327669027

sandeepmistry, 2017-11-02:
Hi @Koepel,

I agree with your suggestion in https://github.com/arduino/Arduino/issues/6652#issuecomment-327676503, would you like to submit a pull request with the change?