arduino/Arduino #6653: code compiles if we write anything after header line #include "pins_arduino.h"cccsdsdasd

ashwanisihag, 2017-08-25:
code complied even we write anything after header line 
#include <avr/wdt.h>cccsdsdasd

per1234, 2017-08-25:
I can't reproduce this with Arduino IDE 1.8.4:
```
C:\Users\per\AppData\Local\Temp\arduino_modified_sketch_482279\sketch_aug25a.ino:1:31: fatal error: e:\arduinoide\arduino-1.8.4\hardware\tools\avr\avr\include\avr\wdt.h>cccsdsdas: Invalid argument
 #include <avr/wdt.h>cccsdsdasd
                               ^
compilation terminated.
exit status 1
Error compiling for board Arduino/Genuino Uno.
```
Which version of the Arduino IDE are you using? Which board are you compiling for?

feikname, 2017-08-25:
I can't reproduce it here either unless I explicitly do `#define cccsdsdasd` before.

However, if I add an space (i.e. `#include <avr/wdt.h> cccsdsdasd` it compiles without problems and produces the following warning:
```text
/tmp/arduino_modified_sketch_261955/sketch_aug25a.ino:1:22: warning: extra tokens at end of #include directive
 #include <avr/wdt.h> cccsdsdasd
```

Which, in my opinion, is the best behavior.

cmaglie, 2017-09-01:
@ashwanisihag I'm closing this as invalid, feel free to comment further if you have any other information to provide.

closed 2017-09-01 by cmaglie