arduino/Arduino #6657: Document bool type in reference

per1234, 2017-08-26:
As discussed in https://github.com/arduino/Arduino/issues/4673, it is desirable to encourage the use of the standard `bool` type rather than Arduino's `boolean` alias. The first step is to document the `bool` type in the Arduino reference.

1. Create a `bool` reference page. Please don't capitalize `bool` in the URL, that was a terrible idea.
2. Copy the documentation from the `boolean` reference page to the `bool` reference page, changing all references to the `boolean` type to `bool`.
3. Replace the documentation in the `boolean` reference page with this:
>boolean is a non-standard type alias for bool defined by Arduino. It's recommended to instead use the standard type [bool](https://www.arduino.cc/en/Reference/bool), which is identical.

Once this has been done I will submit pull requests to remove the use of `boolean` from all official Arduino code and request updates of any other documentation that uses it.

kengdahl, 2018-01-02:
I will happily implement this as soon as there is a response from someone from Responsible  (like @cmaglie) on the subject in https://github.com/arduino/Arduino/issues/4673.

per1234, 2018-01-02:
Now that the reference-en repository is functional I had planned to submit a pull request to add the requested bool reference page, and also propagate that page out to the reference translation repositories after it was merged. However, it would be nice to get a statement on whether this move to encourage `bool` over `boolean` has official approval since otherwise it's unsure if the proposal will actually be able to move forward.

I still maintain my commitment to doing as much of the footwork as is in my power for this project:
- Submit pull requests for IDE bundled and official library examples that use `boolean`.
- Submit pull requests for any reference pages that use `boolean`
- Submit issue reports requesting changes to any documentation using `boolean` that can't be altered via pull requests. I suspect there will not be many required.

kengdahl, 2018-01-02:
I have assigned @cmaglie to this issue besides, just to get a go-ahead.

cmaglie, 2018-01-05:
@per1234 let's go with your plan.

Let's wait about setting the `deprecated` flag, will discuss that in #4673 
It's ok to change the examples from `boolean` to `bool` instead.

closed 2018-01-15 by SimonePDA