arduino/Arduino #6658: Library to Arduino matrix printer

DjamesSuhanko, 2017-08-27:
I did a matrix printer using Arduino and I wrote a library to that. I would like to include it in oficial repository.
The URL to library is https://github.com/DjamesSuhanko/bbPrinter

The article about is http://www.dobitaobyte.com.br/impressora-com-arduino-opencv/

per1234, 2017-08-27:
You need to create a release or tag that matches the `version` value in your library.properties file.

After you have done so, please comment here so we know we can proceed with adding your library to the index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

DjamesSuhanko, 2017-08-27:
Ok, thanks.
I never has used tag, so I did a little research about and if I got it, the thinks are right just now.
I create a tag:
git tag -m '0.1.0'
And I push it:
git push origin 0.1.0
I hope I be right. 
Thank you again!

cmaglie, 2017-08-30:
Done!

closed 2017-08-30 by cmaglie