arduino/Arduino #6659: Reference pages for Serial.write and Serial.print: incorrect return types ?

sterretje, 2017-08-27:
Again not sure if this is in the correct section.

https://www.arduino.cc/en/Serial/Write

[quote]
Returns
byte
write() will return the number of bytes written, though reading that number is optional 
[/quote]
write returns size_t, not byte

https://www.arduino.cc/en/Serial/Print

[quote]
Returns
size_t (long): print() returns the number of bytes written, though reading that number is optional
[/quote]
To my knowledge, size_t on most (or all) Arduinos is an int

per1234, 2017-11-09:
Closed by https://github.com/arduino/reference-en/pull/198

closed 2017-11-09 by per1234

sterretje, 2017-11-09:
[https://cdn.arduino.cc/reference/en/language/functions/communication/serial/write/]
write is still returning a byte.

[https://cdn.arduino.cc/reference/en/language/functions/communication/serial/print/]
print is still returning a size_t(long).

per1234, 2017-11-09:
>write is still returning a byte.

See https://github.com/arduino/reference-en/blob/master/Language/Functions/Communication/Serial/write.adoc

>print is still returning a size_t(long).

See https://github.com/arduino/reference-en/blob/master/Language/Functions/Communication/Serial/print.adoc

The changes aren't immediately pushed to the website but they have been done in the reference repository.

It would be nice if someone in the know from the Arduino team could tell us what the timeframe between repository update and website push (and maybe add that to the reference repository README.md).

sterretje, 2017-11-09:
OK, thanks for that.

Will wait for them to show up (and try to remember to check).