arduino/Arduino #6661: Servo won't turn while accepting commands from an Arduino

Bradfo69, 2017-08-28:
Hello everyone,

I'm asking for your advice -

I'm attempting to turn a servo with my Arduino when it gets orders from a serial contribution, for this situation, my console. I have officially done this with engines and it works fine however when I attempt with my servo it doesn't move by any means. (an example from - http://www.powerjackmotion.com/product-category/electric-motor/servo-motor-range/ ) So far I have this written -

#include <SoftwareServo.h>

SoftwareServo myservo;
int movemotor;
void setup() { 
Serial.begin(9600);
myservo.attach (10);
myservo.write(90);
}
void loop() {
movemotor = Serial.read();
if (movemotor = 111) {
for (int pos = 90; pos >=0; pos--) {
myservo.write(pos);
delay(15);
}
}
else if (movemotor = 99) {
for (int pos = 90; pos <= 180; pos++) {
myservo.write(pos);
delay(15);
}
}
}
I'm powering the servo remotely with a 9-volt battery and have made a point to associate its ground to that of the Arduino, however, can't see whatever else I fouled up.
Any help would be greatly appreciated.
Thanks

feikname, 2017-08-28:
Hello @Bradfo69,

This page is for reporting issues related to the Arduino IDE itself only. To get help with your sketch, please post your problem in the [adequate place](https://github.com/arduino/Arduino/blob/master/CONTRIBUTING.md). Don't worry about this post, it will be simply closed as invalid.

Bradfo69, 2017-08-28:
Thanks for your reply, someone suggests me ask your question here, that's why I'm asking here.

closed 2017-08-28 by per1234

matthijskooijman, 2017-08-29:
@Bradfo69, where did you get the suggestion to ask here? Perhaps you could forward our feedback to them, so they won't forward other people to the wrong place :-)

Bradfo69, 2017-08-29:
@matthijskooijman, I have already forward the feedback to him. Thanks.