arduino/Arduino #6666: Upload blocked by __REMOVE__ text in java script???

chieldejong, 2017-08-29:
Hi all, 
since a few weeks I cannot use my arduino anymore.
After compilation I get the following error message:
--------------------------
Arduino: 1.6.10 (Windows 7), Board: "Arduino/Genuino Uno"

Sketch uses 3,966 bytes (12%) of program storage space. Maximum is 32,256 bytes.
Global variables use 233 bytes (11%) of dynamic memory, leaving 1,815 bytes for local variables. Maximum is 2,048 bytes.
___REMOVE___/bin/avrdude -C___REMOVE___/etc/avrdude.conf -v -patmega328p -carduino -PCOM57 -b115200 -D -Uflash:w:C:\Users\NLY00198\AppData\Local\Temp\build3815d80754a2f55168879c0f4a6e5e2b.tmp/receiverDM4HT.ino.hex:i 
java.io.IOException: Cannot run program "___REMOVE___/bin/avrdude": CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at processing.app.helpers.ProcessUtils.exec(ProcessUtils.java:26)
	at cc.arduino.packages.Uploader.executeUploadCommand(Uploader.java:130)
	at cc.arduino.packages.uploaders.SerialUploader.uploadUsingPreferences(SerialUploader.java:209)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.Sketch.upload(Sketch.java:1187)
	at processing.app.Sketch.exportApplet(Sketch.java:1160)
	at processing.app.Sketch.exportApplet(Sketch.java:1132)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2409)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.io.IOException: CreateProcess error=2, The system cannot find the file specified
	at java.lang.ProcessImpl.create(Native Method)
	at java.lang.ProcessImpl.<init>(ProcessImpl.java:386)
	at java.lang.ProcessImpl.start(ProcessImpl.java:137)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 9 more
An error occurred while uploading the sketch

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
--------------------------
avrdude is in the expected location

removing and reinstalling the most recent IDE (and older version) the problem persists.

Installing Arduino on the PC of a coleage runs fine....

How to get rid of __REMOVE__
regards
ChieldeJong

cmaglie, 2017-08-29:
@chieldejong 
I've already answered to you via email, did you missed it?
Anyway try to delete the "packages" folder inside:

%LOCALAPPDATA%\Arduino15

[image]

and it should be fine.

chieldejong, 2017-08-29:
Hello Christian,
Tx for the mail.
I did not receive a mail before …..

Searching for Arduino15 in explorer did not give any results…
But looking manually following your printscreen resulted in
C:\Users\NLY00198\AppData\Local\Arduino15\packages
Changed the forder name
C:\Users\NLY00198\AppData\Local\Arduino15\packagesRenamed
And now it works again!!!!!!!
Tx a lot

P.S. I found a way around that problem using old fashioned .bat files ;-)

Met vriendelijk groeten/Best Regards

Michiel de Jong
FPP solutions
Philips Research
High Tech Campus 34, room 3.067, 5656 AE  Eindhoven, NL
tel: +31-(0)40-27-92880 Mobile +31 6 12373119
E-mail:Michiel.de.Jong@Philips.com

[image]<https://user-images.githubusercontent.com/423515/29823219-56789958-8cce-11e7-9a74-f0e94fabcbdc.png>

The information contained in this message may be confidential and legally protected under applicable law. The message is intended solely for the addressee(s). If you are not the intended recipient, you are hereby notified that any use, forwarding, dissemination, or reproduction of this message is strictly prohibited and may be unlawful. If you are not the intended recipient, please contact the sender by return e-mail and destroy all copies of the original message.

closed 2017-08-29 by cmaglie