arduino/Arduino #6668: New library: Framework for create a AIO module on NODEMCU using Arduino IDE

bermeom, 2017-08-30:
Hello,

I have created a library for developing AIO modules for our AIO robot using NODEMCU. Could you please add it to Library Manager.

Repository: https://github.com/AIO-Javeriana/AIO-module-nodemcu-arduino

Thanks

per1234, 2017-08-30:
You need to create a release or tag that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

bermeom, 2017-09-01:
Hello,

I already updated my repository.

thanks

per1234, 2017-09-01:
No, there are still no releases or tags in your repository:
https://github.com/AIO-Javeriana/AIO-module-nodemcu-arduino/releases

per1234, 2017-10-03:
@bermeom, still waiting on you to make your library compliant with the Library Manager requirements. If you no longer want to have it added to the Library Manager index then please let us know so we can close this issue.

bermeom, 2017-10-03:
Hello,  I already accepted your pull request, sorry for not responding before.

per1234, 2017-10-03:
[My pull request](https://github.com/AIO-Javeriana/AIO-module-nodemcu-arduino/pull/1) was unrelated to the Library Manager inclusion request you have made here.

Please read this carefully:
**You need to create a release that matches the `version` value in your library.properties file: 0.1.0**
If you won't do that we can't add your library to the Library Manager index.

More information:
- https://help.github.com/articles/creating-releases/
- https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

bermeom, 2017-10-03:
Already !
Thanks,

cmaglie, 2017-10-05:
Done!

closed 2017-10-05 by cmaglie