arduino/Arduino #6671: Can't install the arduino software, some kind of error

MediaCresta, 2017-08-31:
hey there, first i dont speak english very well, so...
I have a problem when trying to install the program, someone has already happened or know how to solve it?.
I will attach the console messages.

/Desktop/Arduino/build/linux/dist$ sudo ./install.sh 
Adding desktop shortcut, menu item and file associations for Arduino IDE...xdg-mime: file '/home/mediacresta/Desktop/Arduino/build/linux/dist/lib/arduino-arduinoide.xml' does not exist
sed: can't read /home/mediacresta/Desktop/Arduino/build/linux/dist/lib/desktop.template: No such file or directory
xdg-icon-resource: file '/home/mediacresta/Desktop/Arduino/build/linux/dist/lib/icons/16x16/apps/arduino.png' does not exist
xdg-icon-resource: file '/home/mediacresta/Desktop/Arduino/build/linux/dist/lib/icons/24x24/apps/arduino.png' does not exist
xdg-icon-resource: file '/home/mediacresta/Desktop/Arduino/build/linux/dist/lib/icons/32x32/apps/arduino.png' does not exist
xdg-icon-resource: file '/home/mediacresta/Desktop/Arduino/build/linux/dist/lib/icons/48x48/apps/arduino.png' does not exist
xdg-icon-resource: file '/home/mediacresta/Desktop/Arduino/build/linux/dist/lib/icons/64x64/apps/arduino.png' does not exist
xdg-icon-resource: file '/home/mediacresta/Desktop/Arduino/build/linux/dist/lib/icons/72x72/apps/arduino.png' does not exist
xdg-icon-resource: file '/home/mediacresta/Desktop/Arduino/build/linux/dist/lib/icons/96x96/apps/arduino.png' does not exist
xdg-icon-resource: file '/home/mediacresta/Desktop/Arduino/build/linux/dist/lib/icons/128x128/apps/arduino.png' does not exist
xdg-icon-resource: file '/home/mediacresta/Desktop/Arduino/build/linux/dist/lib/icons/256x256/apps/arduino.png' does not exist
xdg-mime: file '/home/mediacresta/Desktop/Arduino/build/linux/dist/lib/arduino-arduinoide.xml' does not exist
xdg-icon-resource: file '/home/mediacresta/Desktop/Arduino/build/linux/dist/lib/icons/16x16/apps/arduino.png' does not exist
xdg-icon-resource: file '/home/mediacresta/Desktop/Arduino/build/linux/dist/lib/icons/24x24/apps/arduino.png' does not exist
xdg-icon-resource: file '/home/mediacresta/Desktop/Arduino/build/linux/dist/lib/icons/32x32/apps/arduino.png' does not exist
xdg-icon-resource: file '/home/mediacresta/Desktop/Arduino/build/linux/dist/lib/icons/48x48/apps/arduino.png' does not exist
xdg-icon-resource: file '/home/mediacresta/Desktop/Arduino/build/linux/dist/lib/icons/64x64/apps/arduino.png' does not exist
xdg-icon-resource: file '/home/mediacresta/Desktop/Arduino/build/linux/dist/lib/icons/72x72/apps/arduino.png' does not exist
xdg-icon-resource: file '/home/mediacresta/Desktop/Arduino/build/linux/dist/lib/icons/96x96/apps/arduino.png' does not exist
xdg-icon-resource: file '/home/mediacresta/Desktop/Arduino/build/linux/dist/lib/icons/128x128/apps/arduino.png' does not exist
xdg-icon-resource: file '/home/mediacresta/Desktop/Arduino/build/linux/dist/lib/icons/256x256/apps/arduino.png' does not exist
 done!

matthijskooijman, 2017-08-31:
I am not so sure that `install.sh` is intended to work out of the source tree like that. I suspect it is meant to be run from an unpacked zipfile. I am not sure where / if the build process builds such a zipfile (perhaps that's a manual step afterwards?).

In any case, the install script does not actually copy arduino anywhere, it just installs shortcuts. Since you do not really need those, you can also just run arduino from the build tree, e.g. just run `build/linux/work/arduino`.

cmaglie, 2017-08-31:
As @matthijskooijman pointed out the `install.sh` must be run from a release archive (you can find them here https://www.arduino.cc/en/Main/Software).

If you want to build from source you must run `ant` from the `build` folder and the IDE will be built inside `build/linux/work`. The build process can create the release files by running `ant dist`.

closed 2017-08-31 by cmaglie