arduino/Arduino #6672: RFduino Board compile error

Deaniesaur, 2017-08-31:
Just yesterday. I can compile and upload codes to my RFduino module. However, I dont know if an update came and suddenly this bug appeared. Please help me as soon as possible. Needed for a project.
Tried to uninstall/reinstall arduino IDE and RFduino libraries but nothing fixed it.

Error Messages:

Arduino: 1.8.4 (Windows 10), Board: "RFduino"

Board RFduino (platform RFduino, package RFduino) is unknown

Error compiling for board RFduino.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

matthijskooijman, 2017-08-31:
This issue tracker is about reporting bugs in the Arduino IDE, not for getting help with your own code, projects and third-party boards, so I'm closing this issue. I suggest you ask your question on the Arduino forums or with RFduino instead (and be sure to include more details, such as the sketch you are trying to compile and certainly the full compilation output and error messages, probably with verbose compilation enabled, some info on how you installed the RFDuino board files would be helpful too). From a quick look at the error message, it seems that your RFDuino board was not properly installed somehow.

closed 2017-08-31 by matthijskooijman

Deaniesaur, 2017-08-31:
Full Error Log:

Arduino: 1.8.4 (Windows 10), Board: "RFduino"

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.7.0_x86__mdqgnx93n4wtt\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.7.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\dean_\AppData\Local\Arduino15\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.7.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.7.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\dean_\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.7.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\dean_\OneDrive\Mga Dokumento\Arduino\libraries -fqbn=RFduino:RFduino:RFduino -ide-version=10804 -build-path C:\Users\dean_\AppData\Local\Temp\arduino_build_165037 -warnings=none -build-cache C:\Users\dean_\AppData\Local\Temp\arduino_cache_758650 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\dean_\AppData\Local\Arduino15\packages\RFduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose G:\THESIS\Arduino\RFduino_Test\RFduino_Proto3\RFduino_Proto3.ino
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.7.0_x86__mdqgnx93n4wtt\arduino-builder -compile -logger=machine -hardware C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.7.0_x86__mdqgnx93n4wtt\hardware -hardware C:\Users\dean_\AppData\Local\Arduino15\packages -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.7.0_x86__mdqgnx93n4wtt\tools-builder -tools C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.7.0_x86__mdqgnx93n4wtt\hardware\tools\avr -tools C:\Users\dean_\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.7.0_x86__mdqgnx93n4wtt\libraries -libraries C:\Users\dean_\OneDrive\Mga Dokumento\Arduino\libraries -fqbn=RFduino:RFduino:RFduino -ide-version=10804 -build-path C:\Users\dean_\AppData\Local\Temp\arduino_build_165037 -warnings=none -build-cache C:\Users\dean_\AppData\Local\Temp\arduino_cache_758650 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\dean_\AppData\Local\Arduino15\packages\RFduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose G:\THESIS\Arduino\RFduino_Test\RFduino_Proto3\RFduino_Proto3.ino
Using board 'RFduino' from platform in folder: C:\Users\dean_\AppData\Local\Arduino15\packages\RFduino\hardware\RFduino\2.3.3
Using core 'arduino' from platform in folder: C:\Users\dean_\AppData\Local\Arduino15\packages\RFduino\hardware\RFduino\2.3.3
Detecting libraries used...
"C:\Users\dean_\AppData\Local\Arduino15\packages\RFduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-rtti -fno-exceptions -fno-builtin  -w -x c++ -E -CC -mcpu=cortex-m0 -DF_CPU=16000000 -DARDUINO=10804 -mthumb -D__RFduino__ "-IC:\Users\dean_\AppData\Local\Arduino15\packages\RFduino\hardware\RFduino\2.3.3\cores\arduino" "-IC:\Users\dean_\AppData\Local\Arduino15\packages\RFduino\hardware\RFduino\2.3.3\variants\RFduino" "-IC:\Users\dean_\AppData\Local\Arduino15\packages\RFduino\hardware\RFduino\2.3.3\system/RFduino" "-IC:\Users\dean_\AppData\Local\Arduino15\packages\RFduino\hardware\RFduino\2.3.3\system/RFduino/include" "-IC:\Users\dean_\AppData\Local\Arduino15\packages\RFduino\hardware\RFduino\2.3.3\system/CMSIS/CMSIS/Include" "C:\Users\dean_\AppData\Local\Temp\arduino_build_165037\sketch\RFduino_Proto3.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\dean_\AppData\Local\Arduino15\packages\RFduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-rtti -fno-exceptions -fno-builtin  -w -x c++ -E -CC -mcpu=cortex-m0 -DF_CPU=16000000 -DARDUINO=10804 -mthumb -D__RFduino__ "-IC:\Users\dean_\AppData\Local\Arduino15\packages\RFduino\hardware\RFduino\2.3.3\cores\arduino" "-IC:\Users\dean_\AppData\Local\Arduino15\packages\RFduino\hardware\RFduino\2.3.3\variants\RFduino" "-IC:\Users\dean_\AppData\Local\Arduino15\packages\RFduino\hardware\RFduino\2.3.3\system/RFduino" "-IC:\Users\dean_\AppData\Local\Arduino15\packages\RFduino\hardware\RFduino\2.3.3\system/RFduino/include" "-IC:\Users\dean_\AppData\Local\Arduino15\packages\RFduino\hardware\RFduino\2.3.3\system/CMSIS/CMSIS/Include" "C:\Users\dean_\AppData\Local\Temp\arduino_build_165037\sketch\RFduino_Proto3.ino.cpp" -o "C:\Users\dean_\AppData\Local\Temp\arduino_build_165037\preproc\ctags_target_for_gcc_minus_e.cpp"
arm-none-eabi-g++: error: CreateProcess: No such file or directory

exit status 1
Error compiling for board RFduino.

facchinm, 2017-09-04:
Your installation looks broken; check if `C:\Users\dean_\AppData\Local\Arduino15\packages\RFduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++` file exists and, if it doesn't, please uninstall and reinstall RFDuino core

cmaglie, 2017-09-04:
this has been acknowledge as a bug in the IDE installed from Windows Apps Store: #6673

per1234, 2017-09-04:
The [second ](https://github.com/arduino/Arduino/issues/6672#issuecomment-326220595)output @Deaniesaur posted is #6673. The [first ](https://github.com/arduino/Arduino/issues/6672#issue-254246782)is a completely different issue (similar to https://github.com/arduino/Arduino/issues/5146). That issue is commonly caused by Boards Manager not removing the old version folder when you install a new version of a hardware package. Once you manually delete the old version folder the problem goes away.