arduino/Arduino #6673: Arduino IDE 1.8.4 update: Compile error for 3rd party Boards.

Deaniesaur, 2017-08-31:
Before the update. The installed 3rd Party Board are fully functional.
After the update. Errors in compiling 3rd Party Board codes appeared and can't proceed further.

Error Log:

cmaglie, 2017-08-31:
Which version of the IDE are you running?

cmaglie, 2017-08-31:
I see that you're running the latest from the MSFT AppStore.

OK I guess I found the problem, an update is coming...

Deaniesaur, 2017-08-31:
Sorry for the reply sir. yes I use the version from the Windows App Store. Thank you for the reply sir. Hope that the update fix will help.

cmaglie, 2017-08-31:
I've submitted the update, now it's on the Appstore to publish it (usually it takes between 24/48h).
Let's keep the issue open until the update is rolledout.

matthijskooijman, 2017-08-31:
@cmaglie, what was the issue? Is it something in the Windows build process only? I can't see any commits with changes to fix this?

cmaglie, 2017-08-31:
> Is it something in the Windows build process only? 

yes, it was an error in the package build for the appstore (in particular there is a `windowsStore.txt` file to put inside `lib`) that defines some properties to force the IDE to use a different data folder:

https://github.com/arduino/Arduino/blob/master/arduino-core/src/cc/arduino/os/windows/Win32KnownFolders.java#L74

matthijskooijman, 2017-08-31:
Should such a file (and/or other files needed for building (Windows) packages) not be in this repo?

cmaglie, 2017-09-04:
The app is still stuck in the Windows App Store certification phase, the App Store dashboard says

```
Certification will take up to three business days for this submission.
Note: Once your submission is published, it may take up to 24 hours for customers to see the changes.
```

I've submitted the app on Thu, Aug 31 so today should be the third "business day"...

PaulStoffregen, 2017-09-04:
Today is a USA holiday.

cmaglie, 2017-09-05:
> Today is a USA holiday.

oh... I see...

Anyway now the App should be published, @Deaniesaur may you give it a try?

facchinm, 2017-09-12:
@Deaniesaur did you test the updated version? Can this issue be closed?

Deaniesaur, 2017-09-12:
Yes. The windows app versions works for me now. Thank you

closed 2017-09-12 by Deaniesaur

Deaniesaur, 2017-09-12:
Yes Its working now. Thank you

-- 
 <http://www.dlsu.edu.ph> <https://www.facebook.com/DLSU.Manila.100> 
<http://instagram.com/dlsu> <https://twitter.com/dlsumanila> 
<http://dlsumanila.tumblr.com/> <http://iblog.lasalle.ph/>

DISCLAIMER AND CONFIDENTIALITY NOTICE 
The information contained in this e-mail, including those in its 
attachments, is confidential and intended only for the person(s) or 
entity(ies) to which it is addressed. If you are not an intended recipient, 
you must not read, copy, store, disclose, distribute this message, or act 
in reliance upon the information contained in it. If you received this 
e-mail in error, please contact the sender and delete the material from any 
computer or system. Any views expressed in this message are those of the 
individual sender and may not necessarily reflect the views of De La Salle 
University.

basma37, 2019-05-03:
I have the same error but don’t understand how to solve