arduino/Arduino #6676: Clarification request for https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.6.x-package_index.json-format-specification

jantje, 2017-08-31:
Hi
I'm jantje the builder of the Arduino eclipse plugin named Sloeber and I bumped into a issue where I do not understand the behaviour of the Arduino IDE in regards to selecting the toolchain.
I have this problem with several boards but lets take Open XLR8 from alorium https://raw.githubusercontent.com/AloriumTechnology/Arduino_Boards/master/package_aloriumtech_index.json as an example.
In the package_aloriumtech_index.json it states
```
          "toolsDependencies": [
            {
              "packager": "arduino",
              "name": "avr-gcc",
              "version": "4.8.1-arduino5"
            },
```
Because of this I would assume Arduino IDE to select  avr-gcc version" "4.8.1-arduino5". At least this is what sloeber does and  fails with: `cc1.exe: error: -fno-fat-lto-objects are supported only with linker plugin.`.

However -in my install- Arduino  IDE 1.8.2 selects the toolchain that comes with the IDE.
I think it is strange that a platform provider explicitly states a version for a tool and that this is ignored.
Therefore I think the specification needs a update on how the platform provider can specify a version `{runtime.tools.TOOLNAME-VERSION.path}` for a tool or select the "default" `{runtime.tools.TOOLNAME.path}`. 

Thanks
Jantje

cmaglie, 2017-08-31:
> Therefore I think the specification needs a update on how the platform provider can specify a version {runtime.tools.TOOLNAME-VERSION.path} for a tool or select the "default" {runtime.tools.TOOLNAME.path}.

the IDE already has a feature like that, for example here is what I get if I try to dump the avrdude runtime variables:

```
$ ./arduino --get-pref | grep tools | grep avrdude | sort
runtime.tools.arduino-avrdude-6.3.0-arduino8.path=/home/megabug/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino8
runtime.tools.arduino-avrdude-6.3.0-arduino9.path=/home/megabug/Code/arduino/build/linux/work/hardware/tools/avr
runtime.tools.avrdude-6.3.0-arduino8.path=/home/megabug/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino8
runtime.tools.avrdude-6.3.0-arduino9.path=/home/megabug/Code/arduino/build/linux/work/hardware/tools/avr
runtime.tools.avrdude.path=/home/megabug/Code/arduino/build/linux/work/hardware/tools/avr
$
```

as you can see there are `runtime.tools.avrdude-6.3.0-arduino8.path` and `runtime.tools.avrdude-6.3.0-arduino9.path` each pointing to the corresponding version (the arduino9 version is bundled in the IDE).
There is also a variant in the format `{runtime.tools.PACKAGER-TOOLNAME-VERSION.path}`, in particular `runtime.tools.arduino-avrdude-6.3.0-arduino8.path` and `arduino-avrdude-6.3.0-arduino9.path`.
Finally the `runtime.tools.avrdude.path` just selects the latest version available in this case the version bundled in the IDE.

I didn't looked at the json you posted, it may be that the original json is wrong and it should point to a newer version of gcc?

per1234, 2017-08-31:
>Finally the runtime.tools.avrdude.path just selects the latest version available in this case the version bundled in the IDE.

I have updated https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification with this information.

jantje, 2017-08-31:
@cmaglie 
Thanks for the answer, but the problem is not the possibility to select a different toolchain. I fail to understand how the arduino ide chooses between runtime.tools.avr-gcc.path and runtime.tools.avr-gcc-4.8.1-arduino5.path.

In this case the json specifies avr-gcc version" "4.8.1-arduino5" so I would asume arduino ide takes  runtime.tools.avr-gcc-4.8.1-arduino5.path but arduino ide uses runtime.tools.avr-gcc.path.
So for some reason arduino IDE decides to ignore the version number specified in tools->dependencies in the json.

So I'm asking for a clarification how the json provider and the alternative arduino ide provider can know what needs to be done to route to {runtime.tools.TOOLNAME-VERSION.path} or {runtime.tools.TOOLNAME.path}

cmaglie, 2017-09-01:
`{runtime.tools.TOOLNAME-VERSION.path}` -> path of the tool `TOOLNAME` version `VERSION`
`{runtime.tools.TOOLNAME.path}` -> path of the tool `TOOLNAME` at the latest version available

It may be that the latest version available of a specific tool is different from the version requested in the package json (for example, in your case, the tools bundled in the Arduino IDE are more recent than the tool installed through the package, so `{runtime.tools.avr-gcc.path}` will always point to the tool bundled in the IDE).
If you really want a specific version you should use the notation `{runtime.tools.TOOLNAME-VERSION.path}` in the `platform.txt`.

(On a side note: the error `cc1.exe: error: -fno-fat-lto-objects are supported only with linker plugin.` means that the selected compiler doesn't support LTO, so a newer version of the compiler is required.)

jantje, 2017-09-01:
@cmaglie 
Thanks. This is the info I missed. I updated the page as you can see here: https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.6.x-package_index.json-format-specification/_compare/dd2f36af793a7a08708df6673edea3d2639dd911...8717c3eb8e21490f6dcbbde359437518ec55af5e

> The IDE will take care to **choose** the right flavour based on the `host` value, or fail if a needed flavour is missing.

I guess it was the word "choose" that mislead me into thinking the toolchain version {runtime.tools.avr-gcc.path} points to was based on the json file.

closed 2017-09-01 by jantje