arduino/Arduino #6678: the built in library, keyboard.h doesn't compile right on Ubuntu app-store version

jusrose, 2017-08-31:
this is the error msg
Arduino: 1.6.10 (Linux), Board: "Arduino Leonardo"

/snap/arduino-mhall119/3/hardware/tools/avr/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ar terminated with signal 31 [Bad system call]
exit status 1
Error compiling for board Arduino Leonardo.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

per1234, 2017-08-31:
You are using a version of the Arduino IDE that is outdated by over a year.

It also appears that you are using a "Snap" package. Often the versions of the IDE you get from the package managers have been modified, thus we can't offer support for them.

Please always install the official Arduino IDE downloaded from:
https://www.arduino.cc/en/Main/Software
and only report issues after you have verified they still occur using the [hourly build](https://www.arduino.cc/en/Main/Software).

per1234, 2017-09-11:
@jusrose did you try using the latest version of the official Arduino IDE? If so, did it solve the issue for you?

per1234, 2017-10-03:
Closing due to lack of response from @jusrose.

closed 2017-10-03 by per1234