arduino/Arduino #6679: [Library Manager]: Please add additional OpenBCI Libraries

aj-ptw, 2017-08-31:
Please add the following libraries:
https://github.com/OpenBCI/OpenBCI_32bit_Library
https://github.com/OpenBCI/OpenBCI_Ganglion_Library

per1234, 2017-08-31:
#### OpenBCI/OpenBCI_32bit_Library:
- You need to add a library.properties file, following the Arduino Library Specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format
then create a release that matches the `version` value in library.properties.

#### OpenBCI_Ganglion_Library
- You need to move the library to the root of the repository. It is not allowed to be in the OpenBCI_Ganglion_Library subfolder.

- You need to add a library.properties file, following the Arduino Library Specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

- Then create a release that matches the `version` value in library.properties.

Once you have fixed these issues please comment here so we can proceed with adding your libraries to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

aj-ptw, 2017-08-31:
Okay sorry these are all addressed in an upcoming code release. Do you want to merge and then they will be fixed in the release next week. Otherwise I can wait, you can check out the release candidates for both of these libraries and see there are pre-releases with the proper files.

per1234, 2017-08-31:
Sorry about that. I only looked at your master branch. I believe that as long as the release meets the requirements (which it appears to) it will be added to Library Manager. So we can proceed with adding these libraries to the index.

aj-ptw, 2017-09-02:
@per1234 the libraries have been updated! I would love for you to merge this so I can finish these tutorials before a product launch! So excited to be able to just tell people to go to library manager instead of downloading a zip from GitHub. Thanks!!

per1234, 2017-09-02:
cmaglie is in charge of adding the libraries to the index. I'm just a volunteer trying to help with the administration of the issue tracker and have no ability to do anything beyond reviewing the repositories to make sure they are compliant and labeling the issues. cmaglie comes through and does all the outstanding requests every few days and closes the issues at that time.

aj-ptw, 2017-09-02:
@per1234 okay awesome! thanks so much for your time and effort!

cmaglie, 2017-09-04:
Done!

closed 2017-09-04 by cmaglie