arduino/Arduino #6689: Arduino IDE not working when accidentally I reduced the font size to very small

sundaramprasad, 2017-09-03:
I accidentally reduced the arduino font size. And it has stopped starting. I am not able to open Arduino IDE. I am getting following error when i try to use arduino_debug.exe.

java.lang.IllegalArgumentException: Font size must be > 0
        at org.fife.ui.rtextarea.RTextAreaBase.setFont(RTextAreaBase.java:986)
        at org.fife.ui.rsyntaxtextarea.RSyntaxTextArea.setFont(RSyntaxTextArea.java:2510)
        at processing.app.EditorTab.applyPreferences(EditorTab.java:322)
        at processing.app.EditorTab.<init>(EditorTab.java:110)
        at processing.app.Editor.addTab(Editor.java:1672)
        at processing.app.Editor.createTabs(Editor.java:1645)
        at processing.app.Editor.handleOpenInternal(Editor.java:1944)
        at processing.app.Editor.<init>(Editor.java:340)
        at processing.app.Base.handleOpen(Base.java:833)
        at processing.app.Base.restoreSketches(Base.java:518)
        at processing.app.Base.<init>(Base.java:456)
        at processing.app.Base.main(Base.java:140)

Please help me in getting through this problem.
Thanks a lot!

per1234, 2017-09-03:
Previously reported at: https://github.com/arduino/Arduino/issues/6359#issuecomment-324247440 where you will find a fix described.

This issue has already been fixed in the [hourly build](https://www.arduino.cc/en/Main/Software#hourly) and you can solve the problem by using that version of the Arduino IDE but please be aware that it is primarily intended for beta testing.

closed 2017-09-03 by per1234

supuflounder, 2017-09-16:
Is there any way to know that I have used a keyboard shortcut to set the font size to zero?  More importantly, is there a way to disable this?