arduino/Arduino #6690: MySensors library missing in library manager

benji1304, 2017-09-03:
Same as this previous issue the MySensors lib is no longer available from the Arduino library manager

https://github.com/arduino/Arduino/issues/5569

per1234, 2017-09-03:
I can't reproduce this:
[image]

If the issue still occurs for you, please try this:

1. Click the link at the line following **File > Preferences > More preferences can be edited directly in the file**. This will open the Arduino15 (or similar) folder.
2. Delete all the files that start with library_index. Please be very careful when deleting files on your computer. When in doubt, back up!
3. Sketch > Include Library > Manage libraries... > Filter your search... > MySensors.

Hopefully it will now appear in the search results. Let me know if that solves the issue.

benji1304, 2017-09-03:
@per1234 thanks for the assist. Didn't work, but then I noticed I was running an older version of the software. Uninstalled completely and reinstalled, worked fine.

closed 2017-09-03 by benji1304