arduino/Arduino #6694: Arduino Nano + RFID RC522 + Ethernet shield W5100

peter1a, 2017-09-05:
I have this Arduino Nano, Ethernet shield W5100 and RC522 as card reader. They are sharing pins D13, D12, D11, D10 is SS pin for W5100 D8 is SS pin for RC522. Problem is with pin D12, like it doesn't share it.

For example if I unplug D12 from W5100, RC522 reader works, if I wire it back it doesn't work anymore. W5100 works no matter if RC522 is wired or not, so problem would be somehow with pin D12.

Does anybody have some ideas, what I'm doing wrong right here?

per1234, 2017-09-05:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to post your code, using code tags (`</>` button on the toolbar).

closed 2017-09-05 by per1234

PaulStoffregen, 2017-09-05:
Very old ethernet shields (or Chinese clones, which are still sold) may not properly tri-state MISO.

https://www.dorkbotpdx.org/blog/paul/better_spi_bus_design_in_3_steps