arduino/Arduino #6695: Arduino 1.8.4 IDE Network Ports not seen anymore

hallard, 2017-09-05:
Hi there,
I noticed on V1.8.3 some random issues with network ports and now with 1.8.4 it is still everytime

- so my first check was trying with Web Editor => Works
[image]

- 2nd with Arduino 1.6.13  => Works !!
[image]

- now with Arduino 1.8.4 => Don't Works
[image]

- Bonjour Browser see the services
[image]

It's all on the same computer Windows 10 x64 with firewall disabled 
Any idea ?

facchinm, 2017-09-05:
Hi @hallard ,
could you try launching the IDE using `arduino_debug.exe` and paste here any (error) output coming from `jmdns` ? We updated the library after 1.6.13 so there could be some corner cases we didn't encounter yet. Thanks!

hallard, 2017-09-06:
For sure, here the output, the `using proxy DIRECT` arrive may be 1 minutes after starting appears in console.
[image]

What is strange is that yesterday after I placed this issue network ports appeared may be 30/60 minutes after I launched IDE, so I'm waiting today to see when it will appears again ;-)

hallard, 2017-09-06:
Ok, just to be sure I installed all V1.8.x version on separate folder (zip download) and launched them one by one

- 1.6.13 Works (ports seen immediately)
- 1.8.0 Works (ports seen immediately)
- 1.8.1 Works (ports seen immediately)
- 1.8.2 broken (ports not seen immediately and sometimes seen after some time)
- 1.8.3 broken (ports not seen immediately and sometimes seen after some time)
- 1.8.4 broken (ports not seen immediately and sometimes seen after some time)

So even if it has been worked (sometimes) with 1.8.1+ may be something changed on my desktop and definitively broken all after 1.8.1.
What I'm sure, is that on same computer (today) 1.6.13 and 1.8.1 works immediately and others version does not.
Very strange

facchinm, 2018-12-10:
In the last two years we updated `jmdns` (which is being used to discover network boards) to address security and performance problems in the older versions. The boards are now added/removed based on the information coming from that library, so if a board disappears it is due to mdns reporting the "disconnection". 
To be able to solve the issue we should be able to replicate it, so it would be nice if you could add to the report:
- the OS you are using
- the network topology and routers/AP
- the network boards you see disappearing (Yun / MKR1000/ ESP8266 or whatever)
Thanks!

stonemull, 2018-12-11:
Not much help but just some feedback.

Win 7 64
PC>router ethernet via 2 dlink switches.
router tplink TD-W8980
esp8266 (yet to test with esp32) 

It has been a bit intermittent and usually I needed to restart arduino IDE and network OTA devices would then show up immediately, if I did not restart the IDE then usually they would appear within 30 seconds to 10 minutes after programming, rebooting the device helped at times if it did not show after a few minutes.
At least OTA did work for me till today, though I have not used it for 3 or 4 months. tonight I needed it.
No idea what changed but with 1.8.4 and could no longer see any of my ESP8266 devices that previously worked with OTA at all. No idea why, tried restarting IDE multiple times, tried sketches I know worked previously, I did not see a network connection in 4 hours.
I have installed M5stack but can't think of any other changes that may have broken anything.
Anyway I decided to update and installed 1.8.8 .. no luck at all till I restarted the IDE with the OTA device running and it popped up immediately, now it is showing up but no amount of rebooting or waiting would get it to show previously.

frankygoop, 2018-12-28:
Confirmed comments from Hallard, OTA ports broken from 1.8.2 and later. On IDE 1.8.1 works fine