arduino/Arduino #6696: New library: GUI for Arduino

MikhailNatalenko, 2017-09-05:
Hello,

We have created the library for GUI: handling widgets, events, windows and etc. Would your manager support submodules? 

Repository: https://github.com/libEmGUI/emGUI-arduino

Thanks

per1234, 2017-09-05:
I don't believe submodules are supported. See https://github.com/arduino/Arduino/issues/6385#issuecomment-310406324.

romansavrulin, 2017-09-05:
Will Lib manager prefer binary github releases to just tags and git checkout?

cmaglie, 2017-09-13:
> Will Lib manager prefer binary github releases to just tags and git checkout?

No, because it's more error prone than now.

Also archives generated on github are not guaranteed to be stable (i.e. they can change checksum, for example if "gzip" is updated on the github's servers... already happened in the past and also the reason why the archives are now hosted on arduino.cc servers).

per1234, 2017-09-13:
I added a note to the Library Manager FAQ about submodules not being supported:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ/_compare/04d79210c86309b0bfa80e9faba001ce1d0896fd...46b8656d4fd038d1f2c0d8ce4f5a422d0bd5fa36

I'm guessing a subtree would work. https://www.atlassian.com/blog/git/alternatives-to-git-submodule-git-subtree.

@MikhailNatalenko or @romansavrulin, please comment here once your repository is compliant so we know we can proceed with adding the library to the Library Manager index.

MikhailNatalenko, 2017-09-14:
@per1234 We have switched to subtree. Please check if repo is ready for submit.

per1234, 2017-09-14:
The final thing you need to do is to create a 1.0.0 release or tag. After you have done so, please comment here and we can then proceed to add your library to the Library Manager index.

More information: https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

MikhailNatalenko, 2017-09-14:
@per1234  We have created the release and tag

cmaglie, 2017-09-18:
Done!

closed 2017-09-18 by cmaglie