arduino/Arduino #6699: New library: GSMSim

erdemarslan, 2017-09-05:
Hi,
I wrote a new library for gsm modules. This library tested on SIMCOM modules. Please add this library to the Arduino Library Manager.
https://github.com/erdemarslan/GSMSim
Thanks.

per1234, 2017-09-05:
Your repository is not compliant with the requirements for inclusion in the Library Manager index:
1. Add a library.properties file to the root of the repository, as described [here](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format).
2. Create a release or tag that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

erdemarslan, 2017-09-05:
Sorry. I checked and solved problems. Can you check again?
Thanks.

per1234, 2017-09-06:
You still need to create a 1.0.0 release or tag. It looks like you created a branch named 1.0.0 instead.

erdemarslan, 2017-09-06:
I created release 1.0.19 which is same as in library.properties file. Thanks.

erdemarslan, 2017-09-12:
Hi my friend.
Are you check it again?
Thanks.

per1234, 2017-09-12:
Yes, it's fine now. cmaglie will add it to the Library Manager index on the next run through the outstanding requests.