arduino/Arduino #6702: Selected board/port displays white on white

LightningStalker, 2017-09-06:
Not a major issue, but slightly annoying, as it is a bit difficult to tell what board and port is selected since it is completely unreadable.  Occurs in Linux, running Linux Mint 18.4 Xenial, cinnamon desktop, the same behavior occurs in latest hourly build

facchinm, 2017-09-06:
Hi @LightningStalker ,
could you provide a screenshot? I'm unable to reproduce in my current environment. Thanks!

PaulStoffregen, 2017-09-06:
Colors are correct when I run it on Ubuntu 14.04 (64 bit) and Raspbian "jessie".

LightningStalker, 2017-09-06:
[Here is a screenshot](https://imgur.com/a/iuJ2K)

facchinm, 2017-09-08:
It looks more "white on orange" than white on white. I think it's due to the gtk color theme chosen by your distro; I'd close it as "wontsolve"

closed 2017-09-08 by facchinm

PaulStoffregen, 2017-09-08:
Agreed, this is almost certainly a Mint issue.

However, one possible workaround for @LightningStalker might involve editing the arduino start script.  Deleting "-Dswing.defaultlaf=com.sun.java.swing.plaf.gtk.GTKLookAndFeel" might cause the JRE to revert to some Java default rendering that looks wrong for Mint, but probably doesn't suffer from low contrast color choices.

LightningStalker, 2017-09-08:
@facchinm I thought it was the theme too, but changing themes does not solve the issue.  And what do you mean "white on orange"?  Are you looking where the board is selected next to the little dot?  It definitely looks like white on white to my eyes.

@PaulStoffregen Removing that from the script does make it visible, however the text in the menus becomes really small and difficult to read.
[https://imgur.com/a/sEEYW](http://i.imgur.com/zy8NByu.jpg)

per1234, 2017-09-08:
I believe this is the issue @LightningStalker is reporting:
[image]
The text "Arduino/Genuino Micro" should be displayed in the blank white space I have circled in red.

LightningStalker, 2017-09-08:
Yep that's it. ☺

facchinm, 2017-09-11:
Oh, I see, sorry but I was looking at the highlighted "Leonardo ETH", my bad. 

I had a small review of the code we use to setup the GTK look and feel and we don't explicitly set the "selected item" color, so it's a theme issue indeed (or some setting even more buried in system GTK settings).

I also found an old issue about the same problem on an ancient Mint https://github.com/arduino/Arduino/issues/1540 , they could be doing something nasty with the gtk theme at a distribution level

PaulStoffregen, 2017-09-11:
Just for reference, it looks fine on Ubuntu.

[image]