arduino/Arduino #6703: [boards manager] core reference not found

micooke, 2017-09-06:
Ive found that a contributed core installed using the boards manager is not visible to boards in the sketchbook/hardware folder.

I haven't checked whether the reverse is also true.

This means that if you want to add new variants to an existing core, you are not able to test it until you install it using the boards manager

micooke, 2017-09-06:
Sorry, more clearly - you are not able to test it until you create a package for your variants and install it using the boards manager

facchinm, 2017-09-06:
Hi @micooke ,
you are probably missing one nested directory in  `sketchbook/hardware`.
I'm attaching a "variant" core I use to test webUSB stuff, once extracted you'll get `sketchbook/hardware/webusb/avr/`folder structure which gets recognized by the IDE.
The boards, also, are referencing arduino core via `arduino:arduino` directive in `boards.txt`. Let me know if it works for you!

[webusb.zip](https://github.com/arduino/Arduino/files/1280913/webusb.zip)

micooke, 2017-09-06:
Unfortunately, thats not the issue. If i forgot the additional folder my variant board would not be visible

I have no problem referencing arduino:arduino, i had an issue referencing a third party core sandeepmistry:nRF5.

Arduino barfed that it could not find sandeepmistry:nRF5 when i selected my variant. This went away when i created a package and installed it through boards manager. Im not sure if it complained about third party tools as i could not get to that point.

It seems like arduino:arduino is treated differently to other cores in this regard.

micooke, 2017-09-06:
Hmm, now that i think about it i had the folder order reversed i.e.
' hardware/nRF5/my_variant' when i had them both in tge hardware folder i.e. 'hardware/nRF5/sandeepmistry' it was fine

facchinm, 2017-09-06:
Hmmm, sorry, I'm not getting it :slightly_smiling_face:  Which is your situation? Could you past here the directories nesting and the relevant files? Thanks!

micooke, 2017-09-06:
Im sorry for the confusion, i didnt intend that. Ive tried this on only one machine, but on the two latest Arduino versions. I spent a good half a day hacking away till i resolved it, so hopefully its repeatable and not just a once off. 

Install sandeeps [nRF5 core](https://github.com/sandeepmistry/arduino-nRF5using) the board manager : `https://sandeepmistry.github.io/arduino-nRF5/package_nRF5_boards_index.json`

And copy [my variants](https://github.com/micooke/arduino-nRF5-smartwatches) into hardware/nRF5. Then select one of the new boards and Arduino complained that the core sandeepmistry:NRF5 was not knstalled. But installing my variant via the board manager resolved this `https://micooke.github.io/package_nRF5_smartwatches_index.json`

facchinm, 2017-09-07:
Ok, super, I've been able to reproduce it :smile: 
The problem here are the folder names; if you uninstall your core via board manager, you can check that your (sketchbook) core compiles fine if the folder structure is `sketchbook/hardware/whatever/nRF5/variants/...`. The important piece is **nRF5** part, which must match @sandeepmistry 's core first level name. The board manager automatically handles it since it has a broader knowledge about the various cores/tools interdependencies.
Hope it helps! 
@per1234 where can we document it to make it easily accessible?

micooke, 2017-09-07:
Brilliant! Thanks for looking into it for me.
I know the wiki is a bit of a jumble at the moment, but can I recommend you adding a section here : https://github.com/arduino/Arduino/wiki/Arduino-Hardware-Cores-migration-guide-from-1.0-to-1.6

This is what I used to check the syntax for referencing an external tool.

per1234, 2017-09-07:
I feel that it's already documented here:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification#referencing-another-core-variant-or-tool
>Note that we don't need to specify any architecture since the same architecture of "myboard" is used, so we just say "arduino:arduino" instead of "arduino:avr:arduino".

and here:
https://github.com/arduino/Arduino/wiki/Arduino-Hardware-Cores-migration-guide-from-1.0-to-1.6#step-4-add-tools-for-upload-and-burn-bootloader
>Note that the tool is named arduino:avrdude and not simply avrdude, because the IDE should use the avrdude configuration found in the Arduino's AVR cores (note that the reference only specifies the target vendor and tool name, the architecture is assumed to be the same as the core referencing it -- in this case, avr).

I did a little bit of rewording on the latter to make it more clear.

@micooke, do you think that is sufficient documentation? Regarding the "jumble", improvements are always welcome.

micooke, 2017-09-07:
That's probably fine, I must have stopped reading that section around the tool section because I hadn't read that. 
I guess some of the confusion is that the format is implicit, rather than explicit. I.e. Referencing is maintainer:architecture:core or maintainer:core if you have the same architecture or maintainer:tool. Is this correct?

per1234, 2017-09-08:
`maintainer:architecture:core` is not supported. I suppose the assumption is that cores, tools, and variants will always be architecture specific so there is no reason to support cross-architecture referencing.

micooke, 2017-09-08:
@per1234 - isn't that exactly what arduino:avr:arguing is?

micooke, 2017-09-08:
@per1234 - that is the exact same syntax you references earlier
> https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification#referencing-another-core-variant-or-tool
> 
> Note that we don't need to specify any architecture since the same architecture of "myboard" is used, so we just say "arduino:arduino" instead of "arduino:avr:arduino".

micooke, 2017-09-13:
@per1234 - can you explain the following undocumented syntax inconsistency?
In sandeep's nRF5 core repo i have:
bmp.program.tool=blackmagicprobe
bmp.bootloader.tool=sandeepmistry:blackmagicprobe

* If bootloader.tool is `blackmagicprobe` i get the error `cannot find 'blackmagicprobe'` if i use an variant the references sandeep's repo (all good inside the original repo though)
* If program.tool is `sandeepmistry:blackmagicprobe` i get the error `missing 'program.params.verbose' configuration parameter`.

per1234, 2017-09-13:
It's not fully clear to me what you're attempting to do.

I have read the conversation on your pull request: https://github.com/sandeepmistry/arduino-nRF5/pull/186, which gives some more context but it sounds like you also have your own hardware package that references components of [sandeepmistry/arduino-nRF5](https://github.com/sandeepmistry/arduino-nRF5).

Following the ["From git (for core development)](https://github.com/micooke/arduino-nRF5/tree/master#from-git-for-core-development)" instructions, I installed sandeepmistry/arduino-nRF5 via Boards Manager and then manually installed [your fork](https://github.com/micooke/arduino-nRF5/tree/master) to my sketchbook. I was unable to reproduce the errors you got with either `bmp.program.tool` value. Lacking a clear set of steps to reproduce the error, any prior experience with the nRF5 hardware package, or the hardware in question, it's impossible for me to know why we had different results.

I don't think there would ever be a need to reference a hardware package from within that hardware package so I don't believe you should be doing `bmp.program.tool=sandeepmistry:blackmagicprobe` or `*.program.tool=sandeepmistry:openocd` in your pull request.

None of this seems to be related to an issue with the Arduino IDE so I would request that any further discussion be done elsewhere to avoid subjecting the many subscribers of this repository to more off-topic emails. Perhaps this would be best done in a dedicated issue on your fork, otherwise in the pull request thread. If it is later discovered that there is a verifiable bug with the Arduino IDE then it would be appropriate to report it to this issue tracker.

micooke, 2017-09-13:
You could not reproduce the error because you grabbed the wrong repo, you needed my variants repo https://github.com/micooke/arduino-nRF5-smartwatches, not my fork of sandeep's arduino-nRF5 repo.

I have provided steps already in this thread to reproduce the error, and they were reproduced by facchinm and noted as such in this thread. 
> Install sandeeps [nRF5 core](https://github.com/sandeepmistry/arduino-nRF5using) the board manager : `https://sandeepmistry.github.io/arduino-nRF5/package_nRF5_boards_index.json`
> And copy [my [...]

Since then i have sent a pull request to sandeepmistry number 186 to add blackmagicprobe and resolve issues with Arduino IDE referencing  that. 
In my own variants repo i have made the required changes, so both sandeep and my own repo are ahead of the releases you will get out of the board manager.

I have wasted almost three days in trial and error due to Arduino referencing syntax that is inconsistent and undocumented. Further to this i have pointed out that current documentation uses unsupported syntax. I have documented instructions to reproduce the error which you did not read, and i also documented exactly how i resolved these issues.
Note also that you closed this request prior to trying to reproduce it. Im well aware that you are not interested, so i didnt reopen the request.

I thought you might be interested, and that documenting this might save someone else two or three days of their life. I will bother you no further

per1234, 2017-09-13:
I did read all that and we already resolved it 5 days ago. The original problem was simply that your folder structure was incorrect, as you yourself [said](https://github.com/arduino/Arduino/issues/6703#issuecomment-327458288):

>now that i think about it i had the folder order reversed i.e.
' hardware/nRF5/my_variant'

It has already been explained to you that the architecture folder name of the core and the referenced core must match. This was not the case with your installation as your folder structure had the architecture folder named "arduino-nRF5-smartwatches" while sandeepmistry/nRF5 has the architecture folder name nRF5.

I assumed that since that issue had been resolved that your recent question about the blackmagic reference was regarding an entirely new issue you had encountered, which was reinforced by me discovering you had reported the same exact issue in your pull request to sandeepmistry/nRF5. If you expect me to follow your previous instructions and have a mismatch of the architecture folder name between the active core and the referenced core then of course the reference won't work. That should be obvious.

>I have wasted almost three days in trial and error due to Arduino referencing syntax that is inconsistent and undocumented. Further to this i have pointed out that current documentation uses unsupported syntax.

This is absolutely false. I've already pointed out to you that the reference syntax is clearly documented multiple places in the documentation. If you choose not to read it then there is nothing I can do to help you.

Since you have not been able to point out any legitimate bugs in the Arduino IDE or inaccurate information in the documentation I'm going to lock this thread. In the future please restrict your use of the issue tracker to reporting bugs with the Arduino IDE or feature requests. Support requests such as you've made here should instead be posted to the Arduino Forum: http://forum.arduino.cc/