arduino/Arduino #6707: Arduino IDE can't enter gui main interface.

jeffreyhu3, 2017-09-08:
My PC is windows 10 and install the arduino-1.8.4-windows.exe. When I press Arduino icon, the gui popup and left bottom display initializing packages.... , starting..., then gui is disappear. I remember that I decrease the font size by CTRL+mouse wheel before this issue occur.  I run arduino_debug.exe in the Arduino root folder, get following information:
> "C:\Program Files (x86)\Arduino\arduino_debug.exe"
Looking for library 'user32'
Adding paths from jna.library.path: null
Trying user32.dll
Found library 'user32' at user32.dll
DPI detection failed, fallback to 96 dpi
Looking for library 'shell32'
Adding paths from jna.library.path: null
Trying shell32.dll
Found library 'shell32' at shell32.dll
Looking for library 'Ole32'
Adding paths from jna.library.path: null
Trying Ole32.dll
Found library 'Ole32' at Ole32.dll
Loading configuration...
Initializing packages...
Preparing boards...
Starting...
java.lang.IllegalArgumentException: Font size must be > 0
        at org.fife.ui.rtextarea.RTextAreaBase.setFont(RTextAreaBase.java:986)
        at org.fife.ui.rsyntaxtextarea.RSyntaxTextArea.setFont(RSyntaxTextArea.java:2510)
        at processing.app.EditorTab.applyPreferences(EditorTab.java:322)
        at processing.app.EditorTab.<init>(EditorTab.java:110)
        at processing.app.Editor.addTab(Editor.java:1672)
        at processing.app.Editor.createTabs(Editor.java:1645)
        at processing.app.Editor.handleOpenInternal(Editor.java:1944)
        at processing.app.Editor.<init>(Editor.java:340)
        at processing.app.Base.handleOpen(Base.java:833)
        at processing.app.Base.restoreSketches(Base.java:518)
        at processing.app.Base.<init>(Base.java:456)
        at processing.app.Base.main(Base.java:140)

 I have reinstall the IDE many times, but the didn't solve. Now for my PC only can use the old version arduino-1.6.13. Please help me find how to solve the issue? Thanks.

per1234, 2017-09-08:
Closing as duplicate of https://github.com/arduino/Arduino/issues/6689

We welcome bug reports and feature requests but in the future please take the time to search for existing open and closed issues and pull requests before submitting an issue:
https://github.com/arduino/Arduino/issues?&q=
If you do find the issue has already been reported or a fix has been submitted, you are welcome to give it a "thumbs up" by clicking on the smiley face icon at the top right to show your support. Please only comment on the issue if you have additional information to add.

closed 2017-09-08 by per1234

supuflounder, 2017-09-16:
So why is it that I cannot find the original article when I search for it?  All I get are all the duplicate reports (which suggests that if there are that many duplicates that nobody thought of searching for font size <= 0, which is not surprising.  I searched for the problem, and all I got was a set of articles about 1.6.4 failing with empty json files!)