arduino/Arduino #6708: Can't install Arduino program in Mac OS.

720candy, 2017-09-08:
I can't install Arduino program in my MacBook....At first, I can use this program very well.... suddenly it doesn't work and it show its opening screen like this...... pls help me.........
[image]

facchinm, 2017-09-08:
Hi @720candy ,
can you try to check if the problem solves following these instructions? https://github.com/arduino/Arduino/issues/6359#issuecomment-324247440 .
If it doesn't, try running it from the command line to get the whole output.
- launch a terminal
- drag the arduino app in it
- add `/Contents/MacOS/Arduino` at the end (so you get something like `/Users/yourname/Downloads/Arduino.app/Contents/MacOS/Arduino`
- press Enter to launch it
- copy the error code here

720candy, 2017-09-08:
Thank U command my script! @facchinm 

my error code ------

Last login: Fri Sep  8 22:02:18 on ttys000
parkjungui-MacBook-Pro:~ parkjungyeon$ /Users/parkjungyeon/Desktop/Arduino.app/Contents/MacOS/Arduino ; exit;
java.lang.IllegalArgumentException: Font size must be > 0
	at org.fife.ui.rtextarea.RTextAreaBase.setFont(RTextAreaBase.java:986)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextArea.setFont(RSyntaxTextArea.java:2510)
	at processing.app.EditorTab.applyPreferences(EditorTab.java:322)
	at processing.app.EditorTab.<init>(EditorTab.java:110)
	at processing.app.Editor.addTab(Editor.java:1672)
	at processing.app.Editor.createTabs(Editor.java:1645)
	at processing.app.Editor.handleOpenInternal(Editor.java:1944)
	at processing.app.Editor.<init>(Editor.java:340)
	at processing.app.Base.handleOpen(Base.java:833)
	at processing.app.Base.handleOpen(Base.java:819)
	at processing.app.Base.handleOpen(Base.java:815)
	at processing.app.Base.handleNew(Base.java:754)
	at processing.app.Base.<init>(Base.java:460)
	at processing.app.Base.main(Base.java:140)
logout
Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.

facchinm, 2017-09-08:
Ok, so it's a duplicate of https://github.com/arduino/Arduino/issues/6359#issuecomment-324247440 where you will find a fix described.

You can find the `preferences.txt` file in `/Users/yourusername/Library/Arduino15` folder (`Library` is hidden, you can reach it by selecting `Go to...` on Finder and simply writing `Library` in there)

This issue has already been fixed(5427f94) in the hourly build and you can solve the problem by using that version of the Arduino IDE but please be aware that it is primarily intended for beta testing.

closed 2017-09-08 by facchinm