arduino/Arduino #6712: Issues on Ciao Library page

per1234, 2017-09-08:
https://www.arduino.cc/en/Reference/Ciao

Change:
>Arduino Ciao - from now-on simply "Ciao" - is formally a library to use in your sketches.
Ciao has been designed and developed to be modular and easily configurable.

to:
>Arduino Ciao - from now-on simply "Ciao" has been designed and developed to be modular and easily configurable.

(Awkward and unnecessary wording)

----
Change:
>interact with as much connectors as you can possibly imagine; 

to:
>interact with as many connectors as you can possibly imagine;

----
Change:
>Connectors A Connector is a standalone module

To:
>A Connector is a standalone module

----
Change:
>services and/or applications trought specific protocols.

To:
>services and/or applications **through** specific protocols.

----
Change:
>Main Connectors, like RestServer, Shell, MQTT are pre-installed with Ciao, other connectors

To:
>Main **Connectors** like RestServer, Shell, MQTT are pre-installed with Ciao. **Other** connectors

----
Change:
>you get the library status just writing its name

To:
>you get the library status **by** writing its name

----
All links under **Examples** are broken.

tofrnr, 2017-09-13:
addtionally, 
1) the required hardware for Ciao is not clearly definied , and
2) clicking on the following links does not work at all:

> 
> Examples
> 
>     FileOverwrite? : Open a file in write mode, overwriting it
>     FileRead? : Open a file, reads its contents and shows it via serial
>     MQTTBluemix : Sends via MQTT brightness and temperature information that will be shown graphically in the blueMix IBM system
>     RestClient? : How to implement Rest Client
>     ThingSpeak? : Interaction between the Ciao Library and the Thingspeak Cloud
>     ThingSpeakTalkBackRecv? : Interaction between the Ciao Library and the Thingspeak Cloud
>     ThingSpeakTalkBackSend? : Interaction between the Ciao Library and the Thingspeak Cloud
>     RestServer? : Allows access to the analog and digital pin of the board via rest calls
>     BoardModel? : Get board model, using awk command
>     CPUUsage? : Get the CPU usage every one minute, by reading /proc/stat system file
>     Date? : Get the system current date

per1234, 2017-09-13:
>1. the required hardware (board? shield?) for Ciao is not clearly defined at all, and

That's actually what lead me to the page and caused me to find the other issues I reported here. I was trying to help someone on the Arduino Forum who was trying to use the library with a board it didn't support. It actually is listed under the "Yún devices Library" section of https://www.arduino.cc/en/Reference/Libraries but once you get into the Ciao reference page it's not mentioned anywhere.

>2. clicking on the following links does not work at all:

Yes, I already pointed that out:
>All links under Examples are broken.

It was my primary motivation for opening this issue. The other nit picky stuff was just for a the sake of a complete review while I was at it. But good looking out anyways!

kengdahl, 2018-03-13:
There is only one link under examples now and it works.
The errors in the text is fixed.

closed 2018-03-13 by kengdahl