arduino/Arduino #6713: [Library Manager] Please add egoShield library

uStepper, 2017-09-09:
egoShield library for the uStepper board. Source code:
https://github.com/uStepper/egoShield

Thank you

per1234, 2017-09-09:
The release version must match the `version` value in library.properties at the time of the release. You updated the `version` value in your library.properties to 0.1.0 after you created the 0.1.0 release (https://github.com/uStepper/egoShield/compare/0.1.0...master).

Please resolve this issue and then comment here so we know we can proceed with adding your library to the Library Manager index.

uStepper, 2017-09-10:
Did a new, minor bug fix release.

cmaglie, 2017-09-13:
Library added!

For any future release there is no need to post here, just push a new git tag and the new release is published automatically within an hour.

closed 2017-09-13 by cmaglie