arduino/Arduino #6714: floor() and ceil() return integer values, not double (float)

tofrnr, 2017-09-10:
floor() and ceil() return integer values, not double (float), other than ANSI C math.h
http://www.cplusplus.com/reference/cmath/ceil/

Please make Arduino math functions comply to C standards!

facchinm, 2017-09-11:
Related with https://github.com/arduino/Arduino/issues/6098

cousteaulecommandant, 2017-12-25:
As a workaround, use `(floor)(x)` and `(ceil)(x)`, which will not use the macro but the actual function.

tofrnr, 2017-12-26:
IMO that's stupid to have to use cast functions as worksarounds, because floor() and ceil()  are standard C math functions which are simply messed up by Arduino gibberish.
So finally the Aduino gibberish has to be adjusted to standard C definitions, that's it, nothing else.

cousteaulecommandant, 2017-12-26:
...well, I did say it was a workaround, didn't I? :)
In any case I think Arduino.h will eventually have to be purged from macros and instead use standard C++ functions, or templated functions when no alternative exists.