arduino/Arduino #6716: cannot open the arduino program

yunseok1106, 2017-09-11:
Hello i'm Korean user and today I found a bug in arduino from my mistake. If you minimize your screen,(ctrl+mouse roll) error appears in your screen. And you can't open arduino program anymore. Although i tried to reinstall arduino, i can't opened it.

facchinm, 2017-09-11:
Hi @yunseok1106 ,
the issue has already been reported and fixed, use thishttps://github.com/arduino/Arduino/issues/6708#issuecomment-328103507 workaround to restart the IDE in the meantime. 
Sorry for the inconvenience!

closed 2017-09-11 by facchinm

yunseok1106, 2017-09-11:
thank you but in your comments there are only macOs problem but my computer is window10. and ican't not find the file which you commented. help me

facchinm, 2017-09-11:
On Windows, you can find the file under `C:\Users\YourUserName\AppData\Local\Arduino15`.
Please note that AppData folder is hidden, so you need to enable "Dispaly hidden folders" preference under `Folder Preference`.
Otherwise, you can download the latest nightly with the fix applied (https://www.arduino.cc/download.php?f=/arduino-nightly-windows.zip)