arduino/Arduino #6720: Arduino IDE go creazy with variable B1 and B0

mashanz, 2017-09-12:
I wrote code like this:

`void setup() {
}
void loop() {
}
void something(int A1){
}`
[image]

Click verify and no errors. After a while I got bored with variable A1 and change it to B1. So the code become like this:

`void setup() {
}
void loop() {
}
void something(int B1){
}`

Click verify and got ERROR! -_-"

`exit status 1
expected ',' or '...' before numeric constant`
[image]

tried this with another computer got same result.

whats going on?

closed 2017-09-12 by mashanz

tofrnr, 2017-09-12:
I can confirm this bug - why has this issue been closed without any comment or explanation?

per1234, 2017-09-12:
It's explained here: http://forum.arduino.cc/index.php?topic=499796