arduino/Arduino #6721: #include "arduino.h" being added in inconsistent places

ianahner, 2017-09-14:
I do not have time to chase this issue at the moment, but if I find time I will try to track down the cause and append more info. The issue is this:

Different computers have proven to be inconsistent in where they put the #include arduino.h line.

It is always before setup loop, which is good, but in an example today there was an issue. 

I was supporting two different hardware designs, one using software serial and one using hardware serial. There was therefore the following code above the setup loop:

sudo code:
```
#ifdef IS_DESIGN_1
  //declare software serial
#endif
````

The arduino IDE chose to add the include in the middle so that the following occurred: 

```
#ifdef IS_DESIGN_1
  //declare software serial
  #include "arduino.h"
#endif
````

Due to the fact that the design 1 bit didn't currently apply, the compiler missed the include and spit out a bunch of errors. The problem was resolved by manually including arduino.h, but looking in the tmp file that was generated, I can see that it was still ALSO included inside the #ifdef. This code should be reviewed to avoid this. 

I will work on it if I have time, but meanwhile if anyone more familiar with the way the arduino ide build system functions, perhaps you could guess at why this is happening. I do not think it is unreasonable for #ifdef to exist in the header of a file. I know that is not typical of arduino, but in the embedded world where I typically work that is fairly standard practice.

per1234, 2017-09-14:
Which version of the Arduino IDE are you using on each computer?

ianahner, 2017-09-14:
1.8.4 on both computers. Linux on one machine, Windows 10 on the other. 

The Windows 10 machine was ultimately the one that had the issue, though both seem to be fairly arbitrary in their placement.

per1234, 2017-09-14:
Moved to https://github.com/arduino/arduino-builder/issues/248

This is an arduino-builder issue and thus we should continue the issue there.

closed 2017-09-14 by per1234