arduino/Arduino #6723: which data type and which termination is returned by Serial.readString?

tofrnr, 2017-09-14:
as my issue report has been closed unreasonably without any answer, I want to rephrase my question:

which data type and which termination is returned by Serial.readString?

a char* or a cstring or a member of the String class ?
\0 terminated?
\n terminated?
\n\0 terminated?

per1234, 2017-09-14:
As I have already told you in your previous issue report (https://github.com/arduino/Arduino/issues/6722), the issue of the incorrect documentation of the return type of Serial.readString()  has already been reported at https://github.com/arduino/Arduino/issues/6486.

So this non-duplicate component of this issue is a request from you that the termination of Strings be documented.

per1234, 2017-09-14:
Duplicate of https://github.com/arduino/Arduino/issues/6724

closed 2017-09-14 by per1234

tofrnr, 2017-09-15:
This is not a duplicate, the question is about the termination bytes which are added to the input string by Serial.readString
is it 
\0 terminated?
\n terminated?
\n\0 terminated?

tofrnr, 2017-09-15:
reopen it immediately!

matthijskooijman, 2017-09-15:
@tofrnr, commenting on a closed issue, like you did now, is usually better than re-opening a new issue (as you originally did with this issue), since the latter will not likely earn you patience and will cause people to get annoyed.

As for your original question, let me be a bit more explicit: `readString()` returns a `String` object. The documentation says `string`, which is a capitalization typo, hence your question could be seen as a duplicate to that one.

As for how the returned `String` object is terminated, that's just like any other Arduino String object: It is usually internally \0-terminated, but you don't usually see that externally (e.g. the \0 is not included in `length()`).

matthijskooijman, 2017-09-15:
In any case, this issue tracker is intended for improvements to the software and documentation, not really for questions people have (though those sometimes lead to improvements, of course). I won't be re-opening this issue, but if you (possibly base on my answers) have specific suggestions on where the documentation is lacking *and* how it could be improved, feel free to open up another issue with details.

matthijskooijman, 2017-09-15:
After reading your comment about this in another issue (which isn't quite considered good practice either), I'll humour you and provide one more detail: When you press enter on serial (and the serial monitor is configured to send a CR and/or LF), the line ending will be read by `readString` just line any other character read. `readString` does not stop reading on a newline, it stops reading when some time passed without reading new characters (so if you type two lines fast enough readString might read both lines at the same time, including two line endings, one halfway and one at the end). If you want to read a single line, you might need another approach (though I'm not sure if Serial can do this natively, but I would expect there to be some example(s) for this to be found).