arduino/Arduino #6724: to which data type is Arduino String 100% compatible? char* or string or neither one?

tofrnr, 2017-09-14:
to which data type is Arduino String 100% compatible? char* or string or neither one?
as stated, the documentaion is misleading and confusing with respect to spelling (string vs. String)
https://www.arduino.cc/en/Reference/String
, but it also does not state anywhere if String is a ANSI C char* 
or a C++ string from <cstring>
http://www.cplusplus.com/reference/cstring/

matthijskooijman, 2017-09-14:
It is neither, it is a custom Arduino class, probably introduced because gcc on AVR does not offer `std::string`: https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/WString.h

tofrnr, 2017-09-15:
> re: @per1234 per1234 referenced this issue 15 hours ago 
> What is the termination of a String returned by Serial.readString ? #6723 

that is not true, #6723 is not a duplicate of this topic.
#6723 is about how Seria.redString terminates the input strin after having pressed ENTER, and this question has still not been answered.

It is impudent and intolerable that issues are closed and marked as duplicate falsely weithout being resolved in either way!

matthijskooijman, 2017-09-15:
I'll ignore your previous comment, since that is about another issue (where I also just responded).

As for this issue, I think I've responded to your question, so there's no longer any point in keeping this issue open. If, based on the answers, you have specific suggestions on where the documentation could be improved, feel free to open a new issue.

closed 2017-09-15 by matthijskooijman