arduino/Arduino #6725: Please re-add my library to library manager.

jaean123, 2017-09-15:
I changed my library name from ArduinoCommander to ArduinoBLE so it needs proper updating on the library manager.

Here's the original post:
https://github.com/arduino/Arduino/issues/6476

New link to repository:
https://github.com/jaean123/ArduinoBLE-library

Thanks for helping.

per1234, 2017-09-15:
Actually the library name (as defined by the `name` value in library.properties) remains the same: [Arduino Commander](https://github.com/jaean123/ArduinoBLE-library/blob/master/library.properties#L1).

What you've changed is the repository name, and thus the repository URL as well. I don't know what implications this has for Library Manager but cmaglie will take any action that is necessary on the next run through the outstanding Library Manager inclusion requests.

per1234, 2017-09-15:
I just noticed that in your release and the release branch the `name` value is [ArduinoCommander ](https://github.com/jaean123/ArduinoBLE-library/blob/release/library.properties#L1)but in your master branch the `name` value is [Arduino Commander](https://github.com/jaean123/ArduinoBLE-library/blob/master/library.properties#L1) so it's not clear what your intent is for the library name.

cmaglie, 2017-09-18:
The `name` has always been `Arduino Commander` despite the URL of the repo, so I guess the OP just wants the repository URL changed.

Now I've done it.

@jaean123 please comment further if you need something else.

closed 2017-09-18 by cmaglie