arduino/Arduino #6727: EEPROM.put function zeroing next EEPROM position memory

thalesmaoa, 2017-09-15:
Hi, I was playing around with EEPROM lib, in which I need specific use.

I filled my ATmega328p with ones which gave me (where "i" is the position memory):

```
uint16_t eeValue;
EEPROM.get(i, eeValue);
```

```
i = 1 / 65535
i = 3 / 65535
.
.
.
```

But when I use put command, for exemple, in position 17, it put zero to the next position, which is 19:
```
EEPROM.put(17, 65535);
```

```
.
.
.
i = 11 / 65535
i = 13 / 65535
i = 15 / 65535
i = 17 / 65535
i = 19 / 0
i = 21 / 65535
.
.
.
```

When I use EEPROM.write that doesn't happen.

per1234, 2017-09-15:
I believe this demonstration sketch will make it clear what is happening:
```arduino
#include <EEPROM.h>
byte EEPROMcount = 5;

void setup() {
  Serial.begin(9600);
  while (!Serial) {}

  resetEEPROM();
  Serial.println("EEPROM contents after resetEEPROM()");
  displayEEPROM();

  uint16_t uint16_tVariable = 65535;
  EEPROM.put(0, uint16_tVariable);
  Serial.println("EEPROM contents after EEPROM.put(0, uint16_tVariable)");
  displayEEPROM();
  resetEEPROM();

  EEPROM.put(0, 32000);
  Serial.println("EEPROM contents after EEPROM.put(0, 32000)");
  displayEEPROM();
  resetEEPROM();

  EEPROM.put(0, 65535);
  Serial.println("EEPROM contents after EEPROM.put(0, 65535)");
  displayEEPROM();
  resetEEPROM();

  uint32_t uint32_tVariable = 65535;
  EEPROM.put(0, uint32_tVariable);
  Serial.println("EEPROM contents after EEPROM.put(0, uint32_tVariable)");
  displayEEPROM();
  resetEEPROM();

  EEPROM.put(0, (uint16_t)65535);
  Serial.println("EEPROM contents after  EEPROM.put(0, (uint16_t)65535)");
  displayEEPROM();
  resetEEPROM();
}

void resetEEPROM() {
  for (byte address = 0; address <= EEPROMcount; address++) {
    EEPROM.write(address, 1);
  }
}

void displayEEPROM() {
  for (byte address = 0; address <= EEPROMcount; address++) {
    Serial.print(address);
    Serial.print(":");
    Serial.println(EEPROM.read(address));
  }
}

void loop() {}
```

Output:
```
EEPROM contents after resetEEPROM()
0:1
1:1
2:1
3:1
4:1
5:1
EEPROM contents after EEPROM.put(0, uint16_tVariable)
0:255
1:255
2:1
3:1
4:1
5:1
EEPROM contents after EEPROM.put(0, 32000)
0:0
1:125
2:1
3:1
4:1
5:1
EEPROM contents after EEPROM.put(0, 65535)
0:255
1:255
2:0
3:0
4:1
5:1
EEPROM contents after EEPROM.put(0, uint32_tVariable)
0:255
1:255
2:0
3:0
4:1
5:1
EEPROM contents after  EEPROM.put(0, (uint16_t)65535)
0:255
1:255
2:1
3:1
4:1
5:1
```
What happens is that, since it won't fit in an int16_t, 65535 is treated as a 32 bit type and thus EEPROM.put() writes to 4 bytes.

thalesmaoa, 2017-09-16:
@per1234  Thank you! It is clear now!
I just assumed that it was 16bits, but I forgot the signed int value.
Thank you!

closed 2017-09-16 by per1234