arduino/Arduino #6728: end condition for Serial.readString is ambiguous and eventually undefined

tofrnr, 2017-09-15:
re:
> When you press enter on serial (and the serial monitor is configured to send a CR and/or LF), the line ending will be read by readString just line any other character read. readString does not stop reading on a newline, it stops reading when **_some time passed_** without reading new characters (so if you **_type two lines fast enough_** readString might read both lines at the same time, including two line endings, one halfway and one at the end). If you want to read a single line, you might need another approach (though I'm not sure if Serial can do this natively, but I would expect there to be some example(s) for this to be found). 
(quote by matthijskooijman)

 This behaviour is undefined (what is "some time passed" and what is "type fast enough" ?)

Instead, one would expect readString to work like standard C functions:
gets 
reads a string until ENTER (\n), \n is extracted and \0 is appended

std::getline
getline (istream& is, string& str) without explicite delimiter  works like getline(input, str, input.widen(’\n’)), i.e. the standard delimiter is the endline character \n; \n will be extracted and \0 is appended.

if a user wants to read until delimiters different from ENTER (\n), then it is expected to work instead like
getline (istream& is, string& str, char delim);

For simplification, the first argument 
istream& is
can be dropped because it always reads from the specified Serial, Serial1, Serial2, or softSerial stream.

matthijskooijman, 2017-09-15:
From https://www.arduino.cc/en/Serial/ReadString : 
> Serial.readString() reads characters from the serial buffer into a string. The function terminates if it times out (see setTimeout()).

From https://www.arduino.cc/en/Serial/SetTimeout :

> Serial.setTimeout() sets the maximum milliseconds to wait for serial data when using Serial.readBytesUntil(), Serial.readBytes(), Serial.parseInt() or Serial.parseFloat(). It defaults to 1000 milliseconds.

That seems pretty clear to me? Anything that is unclear about how this works?

That `readString()` is apparently not what you need or want seems a different issue. Perhaps you want to use [`readStringUntil()`](https://www.arduino.cc/en/Serial/ReadStringUntil) instead?

To be honest, I think you should have been able to figure this out yourself. I don't think there's anything particularly wrong with the docs in this regard, I just looked at the Serial reference and noticed this `ReadStringUntil()` function that looked promising.

Again, I'm not sure if there's anything left in this issue, but I'll leave it up to you to close this ticket, or provide more suggestions or other info.

tofrnr, 2017-09-15:
ok, I agree - I admittedly missed that default of 1000ms.
But in the end, this whole Serial read String stuff is  gibberish and hotchpotch, different from anything which applies to standard C functions and syntax.

closed 2017-09-15 by tofrnr