arduino/Arduino #6731: Arduino 1.8.4 crashes before it starts

supuflounder, 2017-09-16:
I figured out how to launch it in a terminal session, and got the following:

Josephs-MacBook-Air:~ flounder$ /Applications/Arduino-3.app/Contents/MacOS/Arduino ; exit;
java.lang.IllegalArgumentException: Font size must be > 0
	at org.fife.ui.rtextarea.RTextAreaBase.setFont(RTextAreaBase.java:986)
	at org.fife.ui.rsyntaxtextarea.RSyntaxTextArea.setFont(RSyntaxTextArea.java:2510)
	at processing.app.EditorTab.applyPreferences(EditorTab.java:322)
	at processing.app.EditorTab.<init>(EditorTab.java:110)
	at processing.app.Editor.addTab(Editor.java:1672)
	at processing.app.Editor.createTabs(Editor.java:1645)
	at processing.app.Editor.handleOpenInternal(Editor.java:1944)
	at processing.app.Editor.<init>(Editor.java:340)
	at processing.app.Base.handleOpen(Base.java:833)
	at processing.app.Base.restoreSketches(Base.java:518)
	at processing.app.Base.<init>(Base.java:456)
	at processing.app.Base.main(Base.java:140)

supuflounder, 2017-09-16:
I managed to figure out (via lots of Google searching) where the preferences.txt file was kept, and here are its contents (I was expecting to see a font height of 0 somewhere, but I don't.  There are only two values set to 0:

editor.divider.size=0
last.sketch.count=0
)

=================Library/Arduino/preferences.txt===================
board=atmega328
browser.linux=mozilla
console=true
console.auto_clear=true
console.error.file=stderr.txt
console.length=500
console.lines=4
console.output.file=stdout.txt
editor.antialias=false
editor.caret.blink=true
editor.divider.size=0
editor.divider.size.windows=2
editor.external=false
editor.font=Monaco,plain,10
editor.font.macosx=Monaco,plain,10
editor.indent=true
editor.invalid=false
editor.keys.alternative_cut_copy_paste=false
editor.keys.alternative_cut_copy_paste.macosx=false
editor.keys.home_and_end_travel_far=true
editor.keys.home_and_end_travel_far.macosx=true
editor.keys.shift_backspace_is_delete=true
editor.languages.current=
editor.tabs.expand=true
editor.tabs.size=2
editor.window.height.default=600
editor.window.height.min=290
editor.window.width.default=500
editor.window.width.min=400
export.applet.separate_jar_files=false
export.application.fullscreen=false
export.application.platform=true
export.application.platform.linux=true
export.application.platform.macosx=true
export.application.platform.windows=true
export.application.stop=true
export.delete_target_folder=true
last.screen.height=720
last.screen.width=1280
last.sketch.count=0
last.sketch0.location=274,54,500,600,457
last.sketch0.path=/Users/flounder/Documents/Arduino/Tests/strandtest/strandtest.ino
platform.auto_file_type_associations=true
preproc.color_datatype=true
preproc.enhanced_casting=true
preproc.imports.list=java.applet.*,java.awt.Dimension,java.awt.Frame,java.awt.event.MouseEvent,java.awt.event.KeyEvent,java.awt.event.FocusEvent,java.awt.Image,java.io.*,java.net.*,java.text.*,java.util.*,java.util.zip.*,java.util.regex.*
preproc.output_parse_tree=false
preproc.save_build_files=false
preproc.substitute_floats=true
preproc.substitute_unicode=true
preproc.web_colors=true
programmer=arduino:avrispmkii
run.display=1
run.options=
run.options.memory=false
run.options.memory.initial=64
run.options.memory.maximum=256
run.present.bgcolor=#666666
run.present.exclusive=true
run.present.exclusive.macosx=true
run.present.stop.color=#cccccc
run.window.bgcolor=#FFFFFF
serial.databits=8
serial.debug_rate=9600
serial.parity=N
serial.port=/dev/cu.usbserial-AM01QLZG
serial.stopbits=1
sketchbook.path=/Users/flounder/Documents/Arduino
target=arduino
update.check=true
update.id=6482898988193433326
update.last=1427863304205
upload.using=bootloader
upload.verify=true

per1234, 2017-09-16:
Closing as duplicate of https://github.com/arduino/Arduino/issues/6689

We welcome bug reports and feature requests but in the future please take the time to search for existing open and closed issues and pull requests before submitting an issue:
https://github.com/arduino/Arduino/issues?&q=
If you do find the issue has already been reported or a fix has been submitted, you are welcome to give it a "thumbs up" by clicking on the smiley face icon at the top right to show your support. Please only comment on the issue if you have additional information to add.

closed 2017-09-16 by per1234

supuflounder, 2017-09-16:
I would suggest adding a line of code before the SetFont that simply demands that the font be set to some reasonable size (like 10) if it happens to be zero, somewhere immediately before RTextAreaBase.java:986

per1234, 2017-09-16:
@supuflounder https://github.com/arduino/Arduino/issues/6707#issuecomment-329954728
>So why is it that I cannot find the original article when I search for it? All I get are all the duplicate reports (which suggests that if there are that many duplicates that nobody thought of searching for font size <= 0, which is not surprising. I searched for the problem, and all I got was a set of articles about 1.6.4 failing with empty json files!)

Probably because you only searched for open issue reports. If you had read the [Contributing Rules](https://github.com/arduino/Arduino/blob/master/CONTRIBUTING.md), as instructed when you created this issue report, you would know that you're supposed to [search open **and closed** issue reports](https://github.com/arduino/Arduino/issues?q=).

>I would suggest adding a line of code before the SetFont that simply demands that the font be set to some reasonable size (like 10) if it happens to be zero, somewhere immediately before RTextAreaBase.java:986

If you had read the [Contributing Rules](https://github.com/arduino/Arduino/blob/master/CONTRIBUTING.md), as instructed when you created this issue report, you would know that you're supposed to verify that the issue still occurs using the [hourly build](http://www.arduino.cc/en/Main/Software#hourly). After testing with the hourly build you would have found that the bug was already fixed. If you had read the issue you duplicated, as I instructed you to in my reply to your first comment, you would know that the bug has already been fixed.

@supuflounder https://github.com/arduino/Arduino/issues/6689#issuecomment-329954813
>Is there any way to know that I have used a keyboard shortcut to set the font size to zero? 

You would see the font get super small of course. You will also see the error message in the console/terminal:
```
java.lang.IllegalArgumentException: Font size must be > 0
```

@supuflounder https://github.com/arduino/Arduino/issues/6689#issuecomment-329954813
>More importantly, is there a way to disable this?

Use Arduino IDE 1.8.3 or the [hourly build](http://www.arduino.cc/en/Main/Software#hourly). Or just don't shrink the font down invisibly small.

@supuflounder please stop spamming the issue tracker about this. Every time someone opens/closes an issue or comments all 870 subscribers to this repository get an email. You have now done this 7 times + 3 replies from me + closing the issue = 11 emails x 870 subscribers = 9570 emails sent out. That's fine when it's a legitimate bug report or feature request but it this case it's just a bunch of spam. The bug has been reported, the bug has been fixed. Case closed, let's move on. If you need support then you should post to the [Arduino Forum](https://github.com/arduino/forum-issues). This issue tracker is only to be used for reporting legitimate bugs and feature requests.