arduino/Arduino #6736: Windows version writes inconsistent line-endings

mklhmnn, 2017-09-17:
I'm using Windows 10 and the Arduino IDEA 1.8.4. While my project's .h file is written with 0x0A (\n) line-endings, the last line contains a 0x0D 0x0A. This causes some other tools to reject this file because of inconsistent line-endings.

cmaglie, 2017-09-18:
May you provide a way step-by-step to reproduce the error?

per1234, 2017-09-18:
The IDE actually does this to all files it creates, not only .h.

Steps to reproduce:
- Save a sketch using the Arduino IDE
- Open the sketch in a program that shows line endings (I'm using Notepad++).

[image]

mklhmnn, 2017-10-14:
I still can reproduce this bug with the 1.9.0 beta version.

facchinm, 2017-10-16:
Hi @mklhmnn ,
I just pushed a proposed fix in 1.9 branch (https://github.com/arduino/Arduino/commit/c96cdd7564e45b30aab255823486c86a9e01ea5b) the build should be ready in a couple of minutes.
Would you mind testing if it solves your problem? The correct build number will be PR-beta1.9-BUILD-11.
Thanks!

mklhmnn, 2017-10-20:
I've tried the download from https://www.arduino.cc/download.php?f=/arduino-nightly-windows.zip (the file-time is 20.10.2017 14:33:36 inside the .zip file, but the bug is still there. If you meant a different build, please let me know where I can download it or how to build Arduino from sources.

per1234, 2017-10-21:
@mklhmnn it's this one:
https://www.arduino.cc/download_handler.php?f=/ide_beta/arduino-1.9.0-beta-windows.zip
They have a beta build as well as an hourly (AKA "nightly") build now.

mklhmnn, 2017-10-21:
Yes, the latest beta build fixes the bug. I was expecting the hourly build be more modern than the beta build, because I assumed the beta build to be updated less often, but both coming from the beta-line.

per1234, 2018-03-19:
@facchinm thanks for your work on this! I confirm that the problem is solved for me in Arduino 1.9.0-beta build 38, Windows 10.

So can this be closed now?

facchinm, 2018-03-19:
I'd prefer to close it when beta gets merged into master, but I know it could become very difficult to keep track of it :slightly_smiling_face:

per1234, 2018-03-19:
OK. I just ran across it today when I rediscovered this bug and thought it might have slipped through the cracks. I didn't realize this was the policy.

If this beta branch system is something that will continue to be used in the future then I think it would be worth adding a new label, something like `Fixed in Beta`. If the beta branch is only a temporary system that will be discontinued after the merge then it's sufficient to just review all issues labeled `IDE 1.9.x Beta` after the merge.

facchinm, 2018-03-19:
Second one :wink: The Beta was not meant to be maintained for such a long time but we still need to close some stuff before merging it

closed 2018-05-02 by facchinm