arduino/Arduino #6738: [Library Manager] Please add QC3Control

vdeconinck, 2017-09-17:
Hi,

Inspired by QC2Control, I modified it to create a library to control Qualcomm Quick Charge 3 chargers.
The code is at https://github.com/vdeconinck/QC3Control
(ref: QC2Control lib was added in issue #6341)

Thanks !

Vincent

per1234, 2017-09-17:
The `version` value in your library.properties file must be updated before you make the release. At the time of the 1.2.0 release, the[ `version` value was set to 1.0.0](https://github.com/vdeconinck/QC3Control/blob/v1.2.0/library.properties#L2). Please either redo the 1.2.0 release or create a new one that is compliant.

Once you have fixed this issue please comment here so we know we can proceed with adding your library to the Library Manager index.

More information: https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

vicnevicne, 2017-09-18:
Hi,
Oops, forgot to bump the version number indeed :-/
I deleted the release and made a new v1.2.0, hope it's ok now.
Thanks and sorry again...

cmaglie, 2017-09-21:
Done!

closed 2017-09-21 by cmaglie

CameronLamont, 2017-10-07:
I have been attempting to use the QC3 library - I may be misunderstanding the code but it appears to require the 3rd DMlow pin within the code? For example  

 if (_DmLoPin == 0) return QC_ERROR_QC2_ONLY;

QCControl.cpp lines 104, 127, 168

There are other references to this value including digitalWrites to it (the 3 dm...V() functions).

With this line in-place it never gets to the QC3 parts of the code.  If I use the 2 arg constructor it sets it to zero (Line 19).  I haven't yet attempted a 3 pin setup. (I was wondering if this was an older version of the code compared to your video and descriptions?).

Thanks for the work you've done on this I'm keen to fire it up!

per1234, 2017-10-07:
@CameronLamont this is not an appropriate place to ask for support for this library.