arduino/Arduino #6745: Please add Andee Library

Annikken, 2017-09-20:
Please add Annikken Andee Library. URL: https://github.com/Annikken/Andee. Thanks!

per1234, 2017-09-20:
You need to create a release or tag that contains your library in its Library Manager compliant form. The library is now compliant but at the time of the most recent release it was not due to missing a library.properties file.

For more information see https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Annikken, 2017-09-21:
Hi,
a new release has been created with library.properties. Sorry for the inconvenience. Thanks!

per1234, 2017-09-21:
The `version` value in library.properties must match the version of the release. The `version` value is still set to 3.1.1 in your 3.1.2 release:
https://github.com/Annikken/Andee/blob/v3.1.2/library.properties#L2

Please re-do the release or create a new one with matching `version` value and then comment here so we know we can proceed.

Annikken, 2017-09-21:
Hi. Sorry again. Version has been updated! Thanks!

cmaglie, 2017-09-21:
Done!

closed 2017-09-21 by cmaglie