arduino/Arduino #6752: [Library Manager] Please add Libraries from stm32duino

fpistm, 2017-09-22:
Hi,
Please add to Library Manager (sensors components, connectivity,...)

1. https://github.com/stm32duino/SPBTLE-RF
2. https://github.com/stm32duino/MX25R6435F
3. https://github.com/stm32duino/VL53L0X
4. https://github.com/stm32duino/LSM6DS3
5. https://github.com/stm32duino/LSM6DS0
6. https://github.com/stm32duino/LSM303AGR
7. https://github.com/stm32duino/LPS25HB
8. https://github.com/stm32duino/HTS221
9. https://github.com/stm32duino/LSM6DSL
10. https://github.com/stm32duino/LIS3MDL
11. https://github.com/stm32duino/LPS22HB

Thanks in advance

cmaglie, 2017-09-29:
All done except:

3. https://github.com/stm32duino/VL53L0X

`VL53L0X (Time-of-Flight)`

because we do not allow parenthesis in the name of the library, to fix this:

- remove the release tag
- update the name in the `library.properties`
- commit & push
- re-create the release tag

please note that a library called `VL53L0X` already exists, so `VL53L0X Time-of-Flight` may be a good name.

10. https://github.com/stm32duino/LIS3MDL

another library with the same name already exists. Please rename following the same instruction above.

fpistm, 2017-09-29:
Hi @cmaglie,

Ok for the parenthesis .
One question, as the other are actually not referenced in the Arduino Library manager? why need to rename them ?

fpistm, 2017-09-29:
Done.
Thanks @cmaglie

cmaglie, 2017-09-29:
> as the other are actually not referenced in the Arduino Library manager? why need to rename them ?

the others are ok, only LIS3MDL and VL53L0X needed a rename.
Now I've added all the libraries, so this issue can be closed.

closed 2017-09-29 by cmaglie